espressif/ESP8266_RTOS_SDK #632: ESP-01 never responds after updating to v3.1

metalalive, 2019-06-15:
Hi , I couldn't get response after updating to ESP8266_RTOS_SDK v3.1 and sending few basic AT commands (e.g. ATE0, AT+GMR ...) to my ESP-01s. 

Before building the firmware from source, I changed something in the menuconfig , described as below :
- python interpreter : python3, ( I cannot use python v2 to execute the esptool.py )
- Default baud rate : 115200.
- Flash size : 1 MB (my ESP-01s includes 1MB flash size)
- make monitor baudrate : 115200

Then I built & programmed the flash of my ESP-01s, it looked OK as shown below :
```
esptool.py v2.4.0
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.4.0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:ff:72:ba
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 10832 bytes to 7197...
Wrote 10832 bytes (7197 compressed) at 0x00000000 in 0.6 seconds (effective 133.9 kbit/s)...
Hash of data verified.
Compressed 221168 bytes to 152491...
Wrote 221168 bytes (152491 compressed) at 0x00010000 in 13.6 seconds (effective 130.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 83...
Wrote 3072 bytes (83 compressed) at 0x00008000 in 0.0 seconds (effective 2206.2 kbit/s)...
Hash of data verified.
```
However ,  I cannot get  any response string after sending few basic AT commands to the ESP-01 in my application.

It  was weird because my ESP-01s worked OK with the pre-built firmware I found & updated from ESP8266 non-OS SDK. (I only can use v2.2.1 of the non-OS SDK, due to flash size issue).

Here is my [sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/3293356/sdkconfig.txt) , could you please help to check what did I do wrong ? thanks for the help.

donghengqaz, 2019-06-27:
If you use "AT" application, you may ask "AT" developers for help and they may come to me for help at my field if neccesary, because I don't know "AT" application.

FayeY, 2019-06-27:
Hi,

The AT firmware based on the 8266 RTOS SDK is the same project as the [esp32-at](https://github.com/espressif/esp32-at).
Change the makefile as 
`
export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP8266
export ESP_AT_MODULE_NAME ?= WROOM-02
`
And then compile the project based on the latest 8266 RTOS SDK. More details are in the doc [How_to_Add_New_Platform_EN.md.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3333972/How_to_Add_New_Platform_EN.md.zip).

If you have any other questions about 8266 RTOS AT, please submit a new issue in the https://github.com/espressif/esp32-at/issues.

Thanks.

closed 2019-06-27 by FayeY