espressif/ESP8266_RTOS_SDK #633: ESP_LOGI uses UART0 although UART1 is set as console output in menuconfig.

SzymonKatra, 2019-06-16:
## Environment
ESP8266_RTOS_SDK Release v3.2
```
git rev-parse --short HEAD
3ad3fd78
```

## Problem Description
I wanted to use UART0 to communicate with other device, so I decided to switch console output 
to UART1 (since I don't need RX) via option in `menuconfig` (UART peripheral to use for console output). After powering on ESP8266, it seemed to work, but I noticed that all logs made by `ESP_LOGI` macro are still printed to UART0. After digging in the source files I found that `s_putchar_func` in `log` component (`log.c`) is not set to the `ets_putc` function. Instead, it uses some generic `putchar` which prints characters to UART0.

### Expected Behavior
Logs are printed to UART1

### Actual Behavior
Logs are printed to UART0.

### My workaround
I solved this problem by using `esp_log_set_putchar(&ets_putc);` just after `app_main` is invoked and successfully got all my logs printed to the UART1. Unfortunately, there still exist some initialization code before `app_main` which uses logging, meaning that few logs are still printed to UART0 on initialization. (i.e. MAC, phy_init, reset_reason).

To bypass this problem I modified `user_init_entry` in `startup.c` to call my function that changes putchar function before all initialization happens.

`startup.c`
```
(...)
/* initialize C++ construture function */
for (func = &__init_array_start; func < &__init_array_end; func++)
    func[0]();

// MY MODIFICATION BEGIN
extern void app_init_hook(void);
app_init_hook();
// MY MODIFICATION END
    
phy_get_bb_evm();

assert(nvs_flash_init() == 0);
assert(wifi_nvs_init() == 0);
assert(rtc_init() == 0);
assert(mac_init() == 0);
assert(base_gpio_init() == 0);
esp_phy_load_cal_and_init(0);
assert(wifi_timer_init() == 0);
(...)
```
and in my `user_main.c`
```
#include <esp_libc.h>
#include <esp_log.h>

void app_init_hook(void)
{
    esp_log_set_putchar(&ets_putc);
}
```

## To summarize, my questions are:
- Why `ets_putc` is not used by default in `log` component?
After setting `UART1` in `menuconfig` I expected that all console output will be printed to UART1 so I can safely use UART0 for my purposes without any collisions (after swapping ports via `uart_enable_swap()`)
- Is there any cleaner way to execute user-defined code before initialization in `user_init_entry` happens?

donghengqaz, 2019-06-27:
I tried just now and find that UART only prints the rom log information not applications, so do you use other specific configuration?

Besides this, you can disable main part of rom log by:
```
[*] Disable ROM UART print
```

sshntt, 2019-07-09:
> I tried just now and find that UART only prints the rom log information not applications, so do you use other specific configuration?
> 
> Besides this, you can disable main part of rom log by:
> 
> ```
> [*] Disable ROM UART print
> ```

Hi @donghengqaz,

I have been trying to achieve the same. I made the changes as suggested in the workaround. I also disabled the ROM UART print from the menuconfig option. Almost all the logs are now coming on UART1, except for the lines below:

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x60000014, len 8, room 16 
tail 8
```

Is there a way to remove these from getting printed to UART0 as well?

Also, the questions asked by @SzymonKatra still remain relevant as I too had to make the changes as per the workaround. 

Thanks

SzymonKatra, 2019-07-09:
@donghengqaz No, I don't use other specific configuration. Thanks for advice with `Disable ROM UART print`. I'll check it out but it won't really give me any improvement, since I swap UART0 pins after initial logs have been printed and ESP_LOGI redirected to the UART1. I solved already encountered problem using the workaround I described. Just wanted to know if it's a bug(?) that `log` component is using `putchar` instead of `esp_putc`. 

@sshntt I think that these logs are printed during bootloader stage and as far as I know it's known drawback.

cranphin, 2019-07-13:
Same question here, I'd like to move all logging to uart1 (as much as possible) so I can use uart0 (which has tx/rx) to talk with an external device :)

donghengqaz, 2019-07-23:
Config the UART1 to be the console tty UART like this：
```
Component config --->
    ESP8266-specific --->
       UART for console output ---> Custom
       UART peripheral to use for console output(0-1) ---> (UART1)
```

And if your project does not work, please use the newest release branch "release/v3.2".

SzymonKatra, 2019-07-29:
@donghengqaz  Waouh, thanks. Seems it's working now. Still getting some output from bootloader (I guess) on startup, but that's not a huge deal. When `app_main` is invoked, all logs are already printed to the UART1, so i can safely swap UART0 ports :wink:

It would be nice to know how to deal with it, but I think the main issue itself is solved now. (maybe old bootloader? `etc Jan 8 2013` doesn't look fresh...)

After resetting device:
**UART0:**
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7332, room 16
0x40100000: _stext at ??:?

tail 4
chksum 0xa3
load 0x3ffe8408, len 24, room 4
tail 4
chksum 0x2a
load 0x3ffe8420, len 3548, room 4
tail 8
chksum 0xd3
csum 0xd3
```   
  
  
**UART1:**
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7332, room 16
0x40100000: _stext at ??:?

tail 4
chksum 0xa3
load 0x3ffe8408, len 24, room 4
tail 4
chksum 0x2a
load 0x3ffe8420, len 3548, room 4
tail 8
chksum 0xd3
csum 0xd3j%BJ*5)$&ISISVS,NKYZy 2nd stage bootloader
I (44) boot: compile time 07:02:43
I (45) qio_mode: Enabling default flash chip QIO
I (53) boot: SPI Speed      : 40MHz
I (59) boot: SPI Mode       : QIO
I (66) boot: SPI Flash Size : 2MB
I (72) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (100) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (112) boot:  2 factory          factory app      00 00 00010000 000f0000
I (123) boot: End of partition table
I (130) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x31308 (201480) map
I (210) esp_image: segment 1: paddr=0x00041320 vaddr=0x3ffe8000 size=0x0051c (  1308) load
I (211) esp_image: segment 2: paddr=0x00041844 vaddr=0x3ffe851c size=0x0019c (   412) load
I (222) esp_image: segment 3: paddr=0x000419e8 vaddr=0x40100000 size=0x05ae0 ( 23264) load
0x40100000: _stext at ??:?

I (243) boot: Loaded app from partition at offset 0x10000
I (280) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (280) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (450) phy_init: phy ver: 1055_12
I (450) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
I (460) test: test
```

My `user_main.c` looks like this:
```
#include "FreeRTOS.h"
#include "driver/uart.h"

void app_main(void)
{
    ESP_LOGI("test", "test");
}

```

SzymonKatra, 2019-08-02:
Discovered finally the real issue I had and opened new one describing it in details - #667 
@donghengqaz Thanks for patience and help,

closed 2019-08-02 by SzymonKatra