espressif/ESP8266_RTOS_SDK #635: Kind request for a WPA2-Enterprise example

d-a-v, 2019-06-17:
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and [have found a similar issue](#442).

## Environment

- Development Kit: wemos/lolin d1 mini
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): v3.2 9f9ee5a9
- Development Env:      make
- Operating System:     linux
- Power Supply:         usb

## Problem Description

related: [#442](https://github.com/espressif/ESP8266_RTOS_SDK/issues/442#issuecomment-451903261)

Would you please be able to provide a WPA2-Enterprise example
like the one [provided in nonos-sdk](https://github.com/espressif/ESP8266_NONOS_SDK/tree/master/examples/wpa2_enterprise
) ?

In nonos-sdk, authentication phase2 TTLS-MSCHAPv2 is working, but not TTLS-PAP.
This authentication scheme is often used, especially in [eduroam](http://eduroam.cstnet.cn) WiFi network.

Thanks

donghengqaz, 2019-06-27:
I am sorry that I had planed to add WPA2-Enterprise to SDK at release/v3.2, but the Wi-Fi developpers have to join another project which is to pass the certification of Wi-Fi specific feature, so the project delay.

When they are free, they will start this work and I will tell you the time when the examples can be supplied.

d-a-v, 2019-09-18:
Was the WiFi certification for the newcomer esp32-s2 ?
Congratulation by the way

d-a-v, 2020-06-05:
Closing per by https://github.com/espressif/ESP8266_RTOS_SDK/commit/95ed4d1e50f23bb89949731ebe06a3b4b8aa7ac0, Thanks !

closed 2020-06-05 by d-a-v