espressif/ESP8266_RTOS_SDK #636: Build  docs/en failed!

c-hamster, 2019-06-19:
I want to build the doc file,but it don't work!

step1:
 make html
/home/esp8266/ESP8266_RTOS_SDK/tools/check_python_dependencies.py -r /home/esp8266/ESP8266_RTOS_SDK/docs/requirements.txt
The following Python requirements are not satisfied:
sphinx>=1.6.5
sphinx-rtd-theme
breathe>=4.7.3
sphinxcontrib-blockdiag>=1.5.3
sphinxcontrib-seqdiag>=0.8.5
sphinxcontrib-actdiag>=0.8.5
sphinxcontrib-nwdiag>=0.9.5
recommonmark
future>=0.16.0 # for ../tools/gen_esp_err_to_name.py
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages. Alternatively, you can run "/usr/bin/python3 -m pip install --user -r /home/esp8266/ESP8266_RTOS_SDK/docs/requirements.txt" for resolving the issue.
**../docs_common.mk:65: recipe for target 'check_python_packages' failed
make: *** [check_python_packages] Error 1**
esp8266@hw-linux:~/ESP8266_RTOS_SDK/docs/en$ 

step2:
pip install --user -r requirements.txt
Collecting sphinx>=1.6.5 (from -r requirements.txt (line 4))
  Using cached https://files.pythonhosted.org/packages/7d/66/a4af242b4348b729b9d46ce5db23943ce9bca7da9bbe2ece60dc27f26420/Sphinx-1.8.5-py2.py3-none-any.whl
Collecting sphinx-rtd-theme (from -r requirements.txt (line 5))
  Using cached https://files.pythonhosted.org/packages/60/b4/4df37087a1d36755e3a3bfd2a30263f358d2dea21938240fa02313d45f51/sphinx_rtd_theme-0.4.3-py2.py3-none-any.whl
Collecting breathe>=4.7.3 (from -r requirements.txt (line 6))
  Using cached https://files.pythonhosted.org/packages/74/83/05c2045d0fc2af9f25d4c7915b78f2109a550c86abe7eb43b3aa36eb5ef0/breathe-4.13.0.post0.tar.gz
    Complete output from command python setup.py egg_info:
    **ERROR: Sphinx requires at least Python 3.5 to run.**
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-oyv7a8/breathe/

step3: my python version is as below!
esp8266@hw-linux:~/ESP8266_RTOS_SDK/docs$ python
python              python2.7           python2-config      python2-pasteurize  python3.6           python3m            
python2             python2.7-config    python2-futurize    python3             python3.6m          python-config  

how to compile the doc file?

FayeY, 2019-06-27:
Could this link help？ https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/?badge=latest

donghengqaz, 2019-06-27:
You should use python3 not python2, so please try "pip3 install --user -r /home/esp8266/ESP8266_RTOS_SDK/docs/requirements.txt". If you have not installed python3, please install it firstly.

closed 2019-07-15 by FayeY