espressif/ESP8266_RTOS_SDK #642: esp_deep_sleep(us): looking for examples

dsptech, 2019-06-24:
Hi,
I'm facing some issue because esp_deep_sleep(us) seems to do nothing (it exit immediately), even enabling the freertos sleep option in menuconfig.
Likely I'm using it in the  wrong manner.
Where I could find working examples related to ESP8266_RTOS_SDK sleep ? 
(google return ESP32 information related only).
Thank you.
Regards.

FayeY, 2019-06-27:
Maybe the example of ESP32 can also work on the ESP8266?

dsptech, 2019-06-27:
Hi @FayeY,
I never used the sleep feature before. 
However if the function match the ESP32 function, as you said, then I missed something because it does not work.
I will update soon.
Thank you.

donghengqaz, 2019-06-27:
It is that the API "esp_deep_sleep" will start a hardware timer to check if current status is OK  and enter to deep sleep mode when everything is OK, so if you really want to enter deep sleep immedietly, you can call API "system_deep_sleep_instant(uint32_t us)".

dsptech, 2019-06-27:
Solved: the connection GPIO16<-->RST was missed.
Now "esp_deep_sleep" work correctly. Also "system_deep_sleep_instant" work fine too.

However, the initial returning of a "not returning" function (ESP32 docs) seems a bit strange. 
@donghengqaz, could the checks inside esp_deep_sleep produce its returning ?
I found no docs about that.
Thank you.

cirias, 2019-07-10:
Hi! I also can't get the `esp_deep_sleep` working. It dumps these when calling this function.
```
Task stack [esp_event_loop_] stack from [0x3fff21a8] to [0x3fff29a4], total [2048] size

                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c 

3fff2880  0x40211550 0x0000017a 0x3fff0228 0x40227b82 0x40211714 0x00000000 0x3ffee520 0x40227bb1 0xffffffff 0x00000000 0x40106614 0x40228b54 0x00000000 0x00000000 0x3fff2910 0x00000000 
0x40227b82: xQueueGenericReset at /home/sirius/github.com/cirias/esp8266_remote_switch/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332

0x40227bb1: prvInitialiseNewQueue at /home/sirius/github.com/cirias/esp8266_remote_switch/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332

0x40228b54: vTaskSetThreadLocalStoragePointerAndDelCallback at /home/sirius/github.com/cirias/esp8266_remote_switch/ESP8266_RTOS_SDK/components/freertos/freertos/tasks.c:4471
```
`system_deep_sleep_instant` works fine though.

pgreenland, 2019-12-25:
Hi All,

I've been having similar problems with the esp_deep_sleep not working.

When called with a reasonable time (30s, expressed as microseconds) it returns immediately.

Interestingly this isn't the case when Wi-Fi is used. If connected and disconnected (I've only tried through the whole esp_wifi_start.....esp_wifi_connect.....esp_wifi_stop sequence) then esp_deep_sleep works fine. Interestingly it still returns (as shown by a log statement after) but goes to sleep correctly.

I had a fiddle with the esp_deep_sleep function when I realised it was unexpectedly returning, though if there's some condition it wont enter sleep because of maybe call it in a loop until its happy. This of course triggered the watchdog but did generate a short stack trace from inside libcore.

`Task watchdog got triggered.

Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Core 0 register dump:
PC      : 0x4024229d  PS      : 0x00000030  A0      : 0x40242209  A1      : 0x3ffeba40  
0x4024229d: wifi_set_opmode_local at ??:?

0x40242209: wifi_set_opmode_local at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60000624  A5      : 0x00079c20  
A6      : 0x00000020  A7      : 0x4010765c  A8      : 0x00000011  A9      : 0x00000090  
A10     : 0x4026f6fb  A11     : 0x00000020  A12     : 0x401061ac  A13     : 0x3ffe9198  
A14     : 0x3ffe9198  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x00000000  

Backtrace: 0x4024229d:0x3ffeba40 0x40243404:0x3ffeba70 0x402420f2:0x3ffeba80 0x40214d3c:0x3ffeba90 Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.
0x4024229d: wifi_set_opmode_local at ??:?

0x40243404: esp_wifi_set_mode at ??:?

0x402420f2: esp_deep_sleep at ??:?

0x40214d3c: app_main`

It appears the Wi-Fi functions are called during the entry to sleep. I'm thinking return value which isn't correctly checked.

Providing a forward declaration for the unexposed system_deep_sleep_instant

`extern void system_deep_sleep_instant(uint32_t us);`

And using this allows the device to enter sleep correctly regardless of Wi-Fi usage.

I have however found that if the wifi has been used it s