espressif/ESP8266_RTOS_SDK #644: esp_timer.h/.c 关于esp_timer_init, esp_timer_stop和esp_timer_delete

woodyfan, 2019-07-02:
根据esp_timer.c发现
`esp_err_t esp_timer_init(void)
{
    return ESP_OK;
}`
那么是否需要在程序的初始化时调用esp_timer_init函数？
在回调函数中调用esp_timer相关函数为防止一个timer被重复create先对该timer 进行stop和delete系统宕机求助
`
esp_timer_stop(user_timer_handler);//problem
esp_timer_delete(user_timer_handler);
user_timer_args.arg = IO_Set->valueint;
esp_timer_create( &user_timer_args, user_timer_handler);
esp_timer_start_once(user_timer_handler, 1000*(uint64_t)IO_time->valueint);
`

woodyfan, 2019-07-02:
assertion "timer" failed: file "esp_timer.c", line 175, function: esp_timer_stop
[0;31mE (141072) ABORT: Error found and abort[0m

donghengqaz, 2019-07-12:
``esp_timer_init`` just is used to make its style like esp-idf, So you can ignore it.

This means `user_timer_handler` is NULL and triggers `assert`.

closed 2019-08-22 by FayeY