espressif/ESP8266_RTOS_SDK #657: 在ESP8266_RTOS_SDK release/v3.2版本上进入深度睡眠，这边实测电流达到100uA以上，与手册上的20uA出入很大

fengfeng0328, 2019-07-19:
在ESP8266_RTOS_SDK release/v3.2版本上进入深度睡眠，这边实测电流达到100uA以上，与手册上的20uA出入很大

donghengqaz, 2019-07-19:
你是怎么测得？是不是还有板级功耗也算进去了？

fengfeng0328, 2019-07-20:
模组ESP-12S，测试程序如下：

/******************************************************************************
 * @ 10秒后系统进入深度睡眠模式，外部唤醒
 * @ EXT_RSTB  管脚拉低，芯片重置并唤醒
*******************************************************************************/
void app_main(void)
{
	for (int timer = 0; timer < 10; timer++) {
		ESP_LOGI(TAG, "timer: %d\n",timer);
		vTaskDelay(1000 / portTICK_RATE_MS);
	}
 
	ESP_LOGI(TAG, "start esp_deep_sleep\n");
 
	/* 发现进入深度睡眠后无法再次烧录固件 */
	esp_deep_sleep(0);
}

用电流表进行模组电流测试，实测电流达到100uA，请问这个问题应该怎样解决？

Eavo, 2019-07-22:
能否麻烦提供下贵司测试的连接图和测试步骤等信息？我们这边来复现一下。

fengfeng0328, 2019-07-22:
问题已经解决，之前测试的模组异常，刚刚换了几个模组测试功耗正常，谢谢支持

closed 2019-07-26 by 27-301