espressif/ESP8266_RTOS_SDK #658: IBM-MQTT status

cranphin, 2019-07-19:
INSTRUCTIONS
============

## Environment

- Development Kit:
- IDF version: 
    78cf0dda
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description

I've been trying to get the ibm-mqtt component to work properly.
I chose it over the esp-mqtt component, because ibm-mqtt is the default in sdkconfig
My code is mostly based on the included example

The problem is I've been running into several problems:

First, the task created by MQTTStartTask with the default config seems to continuously use CPU, which means other tasks with later prio never get a chance to run, which causes many issues:
I have checked this with vTaskGetRunTimeStats

I managed to solve this by setting a non 0 value for CONFIG_MQTT_RECV_CYCLE. This results in the CPU usage being much lower (so most CPU goes to IDLE task, as it should).
But for all the values I tried for this setting (5,50,500), the MQTT handling (publishing messages in my case), becomes very very very very slow, as in seconds or more between messages.

Looking at the ibm-mqtt code, I get the impression it's not written very tidily, the esp-mqtt (which I will try now) looks like it's much better integrated with the workings of ESP-RTOS.

So, I wonder if:
- It would not be better to make esp-mqtt the default in sdkconfig, and why ibm-mqtt is the default.
- If ibm-mqtt will be cleaned up/improved at some point.
- If there is another way to prevent ibm-mqtt from using all CPU, without slowing it down.
- If the default settings for ibm-mqtt are correct.

On request I can create some samples of the exact issues I see, but it would take some time to write them.

Cheers!

cranphin, 2019-07-19:
Oh, right, I know why things became slow. And I think it's bad design in ibm-mqtt :)

MQTTStartTask is called from a task with prio 8, and it copies the prio for the task that creates it (strange design, you would expect it to be a sdkconfig property).
The I call MQTTPublish from a. the thread that created the task (Aslo prio 8), or b. a timer which runs under the timer task (prio 2).
MQTTPublish  tries to take a mutex:
https://github.com/espressif/ESP8266_RTOS_SDK/blob/78cf0dda694102b023015b7673f703d5d2e32ff8/components/mqtt/ibm-mqtt/MQTTClient-C/src/MQTTClient.c#L724

But the mutex is almost always taken by the MQTTTask which runs at higher priority:
https://github.com/espressif/ESP8266_RTOS_SDK/blob/78cf0dda694102b023015b7673f703d5d2e32ff8/components/mqtt/ibm-mqtt/MQTTClient-C/src/MQTTClient.c#L447
It does release it:
https://github.com/espressif/ESP8266_RTOS_SDK/blob/78cf0dda694102b023015b7673f703d5d2e32ff8/components/mqtt/ibm-mqtt/MQTTClient-C/src/MQTTClient.c#L461
But between releasing and re-acquiring the mutex, it doesn't yield, sleep, wait, block, or anything else that would allow another lower priority task to actually get the mutex, so they (almost) never get the mutex to do the MQTTPublish.

cranphin, 2019-07-19:
And thinking about that, the way mutex'es are used are how you would do it on a classic os (windows, linux), where a thread waiting on a mutex will normally get it soon. So I guess the base mqtt code was written for that, and not for RTOS with a priority based scheduler :)

donghengqaz, 2019-07-23:
###   It would not be better to make esp-mqtt the default in sdkconfig, and why ibm-mqtt is the default.

We added the ibm-mqtt much earlier that esp-mqtt, some customers have tried ibm-mqtt, so I set it default only for this reason.

###  If ibm-mqtt will be cleaned up/improved at some point.

We may remove ibm-mqtt at version 4.0 and only esp-mqtt is left.

###    If there is another way to prevent ibm-mqtt from using all CPU, without slowing it down.

I see you have found the key to the problem, it is not a good design to all operation system. I would like to say that ibm-mqtt should be used only one thread, send and recv.

###   If the default settings for ibm-mqtt are correct.

The coder who added this project had tried some feature, but may not be thoughtful. Do not tried ibm-mqtt again.

cranphin, 2019-07-23:
Hi!
Thank your for the reply, it matches my expectations :)
Based on it and my own findings I'll switch to and use esp-mqtt now.
I'll close the issue :)

closed 2019-07-23 by cranphin