espressif/ESP8266_RTOS_SDK #663: Using xEventGroupSetBitsFromISR gives undefined reference to `xTimerPendFunctionCallFromISR'

cranphin, 2019-07-24:
## Environment

- Development Kit: 'NodeMCU' esp8266 board
- IDF version: 34f7d836
- Development Env: Make
- Operating System: Ubuntu
- Power Supply: USB

## Problem Description

I wrote this code, which is pretty much copy paste from the documentation of the xEventGroupSetBitsFromISR method:
```
	/* xHigherPriorityTaskWoken must be initialised to pdFALSE. */
	BaseType_t xHigherPriorityTaskWoken = pdFALSE;

	const BaseType_t xResult = xEventGroupSetBitsFromISR( oneWireEventGroup,  ONEWIRE_CONTINUE_BIT, &xHigherPriorityTaskWoken );

	/* Was the message posted successfully? */
	if( xResult != pdFAIL ) {
		/* If xHigherPriorityTaskWoken is now set to pdTRUE then a context
		switch should be requested.  The macro used is port specific and will
		be either portYIELD_FROM_ISR() or portEND_SWITCHING_ISR() - refer to
		the documentation page for the port being used. */
		portEND_SWITCHING_ISR( xHigherPriorityTaskWoken );
	}
```
This results in the following compiler error:
```
/esp/ESP8266_RTOS_SDK/components/esptool_py/esptool /esp/project
patching file esptool.py
/esp/project
Toolchain path: /esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-92-g8facf4c
Compiler version: 5.2.0
esptool.py v2.4.0
CXX build/main/1wire.o
AR build/main/libmain.a
Python requirements from /esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
LD build/NewSens.elf
/esp/project/build/main/libmain.a(1wire.o):(.literal._Z17hw_timer_callbackPv+0x8): undefined reference to `xTimerPendFunctionCallFromISR'
/esp/project/build/main/libmain.a(1wire.o): In function `hw_timer_callback(void*)':
/esp/project/main/1wire.cpp:276: undefined reference to `xTimerPendFunctionCallFromISR'
collect2: error: ld returned 1 exit status
/esp/ESP8266_RTOS_SDK/make/project.mk:447: recipe for target '/esp/project/build/NewSens.elf' failed
make: *** [/esp/project/build/NewSens.elf] Error 1
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```
I think that maybe INCLUDE_xTimerPendFunctionCall  misses in FreeRTOSConfig.h in the SDK?

I didn't see an option in sdkconfig to enable this

donghengqaz, 2019-07-26:
There is no "INCLUDE_xTimerPendFunctionCall" at SDK. So could you add "INCLUDE_xTimerPendFunctionCall" to FreeRTOSConfig.h and I will add this one to "menuconfig" to enable some function later.

Fonger, 2019-09-07:
@donghengqaz I need this config too. What is the ETA of this feature in menuconfig?

dpmcgabriel, 2020-04-29:
There was any solution for this case? Trying to use xTimerPendFunctionCallFromISR, but having  implicit declaration of function 'xTimerPendFunctionCallFromISR'; error