espressif/ESP8266_RTOS_SDK #664: esp-mqtt经常收到空数据，然后模块自动重启了

sonwa, 2019-07-25:
请问是什么问题？

TOPIC=devices/a662f7875ba2502b8c8156205344ca52
DATA=ffff01001801
I (16647) MQTT_CLIENT: deliver_publish, message_length_read=1024, message_length=92
I (16657) MQTT_EXAMPLE: MQTT_EVENT_DATA
TOPIC=devices/a662f7875ba2502b8c8156205344ca52
DATA=ffff01001801
I (18227) MQTT_CLIENT: deliver_publish, message_length_read=1024, message_length=57
I (18237) MQTT_EXAMPLE: MQTT_EVENT_DATA
TOPIC=<null>
DATA=
I (18237) MQTT_CLIENT: deliver_publish, message_length_read=528, message_length=51
I (18257) MQTT_EXAMPLE: MQTT_EVENT_DATA
TOPIC=<null>
DATA=
E (18267) MQTT_CLIENT: Read error or end of stream
I (18277) MQTT_CLIENT: Reconnect after 5000 ms
I (18277) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (25817) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (25827) MQTT_EXAMPLE: MQTT_EVENT_CONNECTED
I (25827) MQTT_EXAMPLE: sent publish successful, msg_id=495
I (25847) MQTT_CLIENT: deliver_publish, message_length_read=19, message_length=19
I (25857) MQTT_EXAMPLE: MQTT_EVENT_DATA

donghengqaz, 2019-07-26:
你用的是什么 example，或者你能不能提供你的 example 和详细的信息给我，你直接贴几句 log 我也没办法看啊。

ustccw, 2019-08-02:
@sonwa 妨请按照 https://www.espressif.com/zh-hans/company/contact-extra/technical-inquiries-software 上报问题, 方便我们复现, 定位, 解决.

FayeY, 2019-08-22:
由于长时间无回复，此问题暂时关闭。如有需要，可以重新 reopen issue 继续讨论。谢谢。

closed 2019-08-22 by FayeY