espressif/ESP8266_RTOS_SDK #666: esp_http_client GET 访问失败的问题

TAOGEDE, 2019-08-01:
您好我希望使用 ESP8266 通过HTTP 的GET来访问服务器的资源，所以我测试了位于 ESP8266_RTOS_SDK/examples/protocols/esp_http_client/ 的demo 

已下是代码：
esp_http_client_config_t config = {
        .url = "http://weixin.taogede.com/test",   
        .event_handler = _http_event_handler,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);

    // GET
    esp_err_t err = esp_http_client_perform(client);
    if (err == ESP_OK) {
        ESP_LOGI(TAG, "HTTP GET Status = %d, content_length = %d",
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
    } else {
        ESP_LOGE(TAG, "HTTP GET request failed: %d", err);
    }

打印信息：HTTP_CLIENT: HTTP GET Status = 200, content_length = -1
而demo默认的url:"http://httpbin.org/get"经过测试是可以的访问成功的

经过测试我发现，TP5构架返回的HTTP报文的首部是没有包含 Content-Length
而 http://httpbin.org/get 返回的HTTP报文的首部含有Content-Length

-------------------------------http://httpbin.org/get 返回的HTTP报文--------------------------------
HTTP/1.1 200 OK
Access-Control-Allow-Credentials: true
Access-Control-Allow-Origin: *
Content-Type: application/json
Date: Thu, 01 Aug 2019 07:09:01 GMT
Referrer-Policy: no-referrer-when-downgrade
Server: nginx
X-Content-Type-Options: nosniff
X-Frame-Options: DENY
X-XSS-Protection: 1; mode=block
Content-Length: 150
Connection: Close

{
  "args": {}, 
  "headers": {
    "Host": "httpbin.org"
  }, 
  "origin": "113.227.228.214, 113.227.228.214", 
  "url": "https://httpbin.org/get"
}
----------------------------------------------------------------------------------------------------------
--------------------------http://weixin.taogede.com/test 返回的HTTP报文---------------------------
HTTP/1.1 200 OK
Server: nginx
Date: Thu, 01 Aug 2019 07:04:54 GMT
Content-Type: text/html; charset=utf-8
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding
X-Powered-By: PHP/7.3.3

4
test
0
----------------------------------------------------------------------------------------------------------

另外我的服务器使用的是TP5构架

SDK 版本 IDF3.1

ustccw, 2019-08-01:
@TAOGEDE 发送 GET 前试一下 `esp_http_client_set_header(client,"content-type","application/json");`

TAOGEDE, 2019-08-01:
@ustccw 我使用TCP调试助手发送

GET /test HTTP/1.1
Host:weixin.taogede.com
Connection:close

能够收到服务器返回的报文

ustccw, 2019-08-01:
@TAOGEDE 发送 GET 前试一下 esp_http_client_set_header(client,"content-type","application/json");

TAOGEDE, 2019-08-01:
@ustccw 试了不好用呀兄弟

ustccw, 2019-08-01:
@TAOGEDE 
1. 服务器返回 `200`, 但 `Content-length` 是 `-1`, 大可能是服务器返回 `chunked` 数据.  
为了不使服务器返回 `chunked` 数据, 需要为期望收到的数据设置 `Content-Type`, 具体 `type` 可参考 `http://tool.oschina.net/commons/`

2. 可以在电脑端通过 `wireshark` 抓包, 查看 `GET` 请求 `HTTP header`,  `ESP8266` 通过 `esp_http_client.h` 中接口, 构造和 `PC` 同样的 `HTTP header`

3. 可以在 `make menuconfig` -> `Component config` -> `Log output` -> `Default log verbosity` 设置 `Debug`, 查看出错原因.

4. 如果服务器只能回复 `chunked` 数据, 那就应用层按照 `chunked` 数据格式解析, 读取, 而不通过 `Content-Length`.

TAOGEDE, 2019-08-01:
可以了

TAOGEDE, 2019-08-01:
@ustccw 谢谢

closed 2019-08-01 by TAOGEDE

ustccw, 2019-08-01:
@TAOGEDE 如何解决的呢, share 一下？

TAOGEDE, 2019-08-02:
@ustccw 
为了节省内存我使用 esp_http_client_get_content_length(client) 获得GET的到的数据的长度，然后使用 malloc() 申请内存，最后使用 esp_http_client_read() 读出数据。但是正如上面提到的问题当服务器分块发送数据时 esp_http_client_get_content_length(client) 似乎不能正确的获取到数据长度。（于是我就以为esp_http_client_read()不能读出数据了）

你的回答提醒了我，可能 esp_http_client_read() 仍然可以读出数据，所以我申请了固定大小的内存使用 esp_http_client_read() 成功的读出了数据。
有趣的是使用 esp_http_client_read() 读出的数据是不含有 chunked分块数据大小信息的，使用 esp_http_client_read() 可以直接读出服务器分块发送的数据而不需要在做其他处理

最后再次感谢你

希望乐鑫能解决esp_http_client_get_content_length() 的问题

ustccw, 2019-08-02:
@TAOGEDE 感谢分享.