espressif/ESP8266_RTOS_SDK #667: Enabling Espressif VFS breaks the output redirection to UART1

SzymonKatra, 2019-08-02:
## Environment

- Development Kit: LoLin new nodemcu v3
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 8dbb0c67
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description

The problem is related to the #633 
I followed the instructions from @donghengqaz how to redirect all output to UART1 and finally made it working on simple example project. However, I was not able to achieve it on my personal project, so I started to look for differences in `sdkconfig`. I discovered that enabling **Espressif VFS** causes system to behave weird, because some logs were printed to UART0 and some to UART1.

### Expected Behavior
No logs should be printed to UART0 (except bootloader messages). Everything should be logged to UART1.

### Actual Behavior
Logs start to print to UART1, but durning initialization stage output is changed to UART0. Logs until 228 ms are printed to UART1 and all logs later are printed to UART0.

### Steps to repropduce

I prepared minimal example which reproduces this problem:
- Output UART1 and Disabled VFS - [uart_problem_disabled_vfs.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3462609/uart_problem_disabled_vfs.zip)
- Output UART1 and Enabled VFS - [uart_problem_enabled_vfs.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3462610/uart_problem_enabled_vfs.zip)

1. Connect UART0 TX output to serial-usb converter, let's say it's `/dev/ttyUSB0`
2. Connect UART1 TX output to serial-usb converter, let's say it's `/dev/ttyUSB1`
3. Read from both ports simulatenously using `idf_monitor`

Application code is just a simple `printf` from the `get-started` example

### Code to reproduce this issue
```
#include <stdio.h>
#include "esp_system.h"

void app_main(void)
{
    printf("SDK version:%s\n", esp_get_idf_version());
}
```

`sdkconfig` in these project differs only in `Virtual file system` section
**`uart_problem_disabled_vfs`:**
```
#
# Virtual file system
#
CONFIG_USING_ESP_VFS=
```

**`uart_problem_enabled_vfs`:**
```
#
# Virtual file system
#
CONFIG_USING_ESP_VFS=y
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
```

Redirection to UART1 was done using following options:
```
CONFIG_CONSOLE_UART_DEFAULT=
CONFIG_CONSOLE_UART_CUSTOM=y
CONFIG_CONSOLE_UART_NONE=
CONFIG_CONSOLE_UART_CUSTOM_NUM_0=
CONFIG_CONSOLE_UART_CUSTOM_NUM_1=y
CONFIG_CONSOLE_UART_NUM=1
CONFIG_CONSOLE_UART_BAUDRATE=74880
CONFIG_UART0_SWAP_IO=
CONFIG_DISABLE_ROM_UART_PRINT=y
```

## Debug Logs

### Debug logs from `uart_problem_disabled_vfs` (correct behaviour)
#### UART0 (`/dev/ttyUSB0`)
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x60000014, len 8, room 16
tail 8
```

#### UART1 (`/dev/ttyUSB1`)
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x60000014, len 8, room 16
tail 8I (27) boot: ESP-IDF v3.2-8-g8dbb0c67 2nd stage bootloader
I (28) boot: compile time 19:59:36
I (28) qio_mode: Enabling default flash chip QIO
I (35) boot: SPI Speed      : 40MHz
I (42) boot: SPI Mode       : QIO
I (48) boot: SPI Flash Size : 2MB
I (54) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  2 factory          factory app      00 00 00010000 000f0000
I (105) boot: End of partition table
I (112) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x3131c (201500) map
I (192) esp_image: segment 1: paddr=0x00041334 vaddr=0x3ffe8000 size=0x0051c (  1308) load
I (193) esp_image: segment 2: paddr=0x00041858 vaddr=0x3ffe851c size=0x0019c (   412) load
I (204) esp_image: segment 3: paddr=0x000419fc vaddr=0x40100000 size=0x05ae0 ( 23264) load
0x40100000: _stext at ??:?

I (225) boot: Loaded app from partition at offset 0x10000
I (261) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (261) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (441) phy_init: phy ver: 1055_12
I (451) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
SDK version:v3.2-8-g8dbb0c67
```

### Debug logs from `uart_problem_enabled_vfs` (incorrect, weird behaviour)
#### UART0 (`/dev/ttyUSB0`)
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x60000014, len 8, room 16
tail 8I (426) phy_init: phy ver: 1055_12
I (426) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
SDK version:v3.2-8-g8dbb0c67
```

#### UART1 (`/dev/ttyUSB1`)
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x60000014, len 8, room 16
tail 8I (27) boot: ESP-IDF v3.2-8-g8dbb0c67 2nd stage bootloader
I (28) boot: compile time 19:59:36
I (28) qio_mode: Enabling default flash chip QIO
I (35) boot: SPI Speed      : 40MHz
I (42) boot: SPI Mode       : QIO
I (48) boot: SPI Flash Size : 2MB
I (54) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  2 factory          factory app      00 00 00010000 000f0000
I (105) boot: End of partition table
I (112) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x33954 (211284) map
I (196) esp_image: segment 1: paddr=0x0004396c vaddr=0x3ffe8000 size=0x00604 (  1540) load
I (197) esp_image: segment 2: paddr=0x00043f78 vaddr=0x3ffe8604 size=0x0019c (   412) load
I (207) esp_image: segment 3: paddr=0x0004411c vaddr=0x40100000 size=0x05ae0 ( 23264) load
0x40100000: _stext at ??:?

I (228) boot: Loaded app from partition at offset 0x10000
```

### Summary
As we can see some last logs are printed to UART0 instead of continuing on UART`
```
I (426) phy_init: phy ver: 1055_12
I (426) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
SDK version:v3.2-8-g8dbb0c67
```
And some of them are even lost!
```
I (261) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
```

## Workaround
I described workaround in #633, but still didn't get answer to my question - Why `log` component is using `putchar` instead of `esp_putc`?

xuhongv, 2019-08-03:
- Here is a Chinese version of the blog tutorial, I wonder if it is suitable for you? https://blog.csdn.net/xh870189248/article/details/97286771

xuhongv, 2019-08-03:
- methods 1：

```
#define BUF_SIZE (1024)

static void echo_task()
{

    uart_enable_swap(); //交换

    uart_config_t uart_config = {
        .baud_rate = 9600,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE};

    uart_param_config(UART_NUM_0, &uart_config);
    uart_driver_install(UART_NUM_0, BUF_SIZE * 2, 0, 0, NULL);

    // Configure a temporary buffer for the incoming data
    uint8_t *data = (uint8_t *) malloc(BUF_SIZE);

    while (1) {

        // 读取数据
        int len = uart_read_bytes(UART_NUM_0, data, BUF_SIZE, 20 / portTICK_RATE_MS);

        if (!len)
            continue;
            
        // Write data back to the UART
        uart_write_bytes(UART_NUM_0, (const char *) data, len);
    }
}

```

SzymonKatra, 2019-08-03:
@xuhongv Thanks for the answer, but I don't see how it's related to the problem I desribed. I already figured out how to swap UART0 ports and it's certainly not a subject of this issue.

donghengqaz, 2019-08-05:
Thanks for your report, it is a bug that VFS does't map UART1 to stdout/stderr, so the UART0 prints the application log also. New I have added this feature to release/v3.2 and master and then it may fix your problem.

closed 2019-08-22 by FayeY

JackieLi1011, 2019-12-18:
Hi, I m facing the same problem you mentioned before.I tried to disable the VFS in sdkconfig but I found this selection cannot be disabled.And I'v tried all the method above except this one.