espressif/ESP8266_RTOS_SDK #672: RTOS_SDK 3.2 如何使用RTC计时功能

match0131, 2019-08-08:
请问3.2SDK如何使用RTC计时功能，以前2.x.x版本的下列接口能否在新版SDK中提供？
system_get_rtc_time
system_rtc_clock_cali_proc
system_rtc_mem_write
system_rtc_mem_read

thisdream, 2019-08-12:
同问

donghengqaz, 2019-08-12:
```
system_get_rtc_time
system_rtc_clock_cali_proc
```
这 2 个接口还在，应该可以使用。

```
system_rtc_mem_write
system_rtc_mem_read
```
不建议再使用，建议使用 RTC_DATA_ATTR 去实现相关功能，使用这个宏申明的变量会被链接到 RTC memory。

thisdream, 2019-08-12:
那我该怎样使用rtc计时呢

donghengqaz, 2019-08-20:
你可以暂时用 `system_get_rtc_time` 来计时，最好是通过 SNTP 来获取准确时间，因为 RTC 用的是内部 RC 振荡器，精度不是太好。

thisdream, 2019-08-22:
您能在3.2里使用system_get_rtc_time吗?我没找到啊，请问system_get_rtc_time在哪个头文件里？

Fonger, 2019-09-07:
@thisdream

You can simply declare the function with `extern` keyword.
```c
extern uint32_t system_get_rtc_time(void);
void app_main() {
   printf("RTC_time: %d\n", system_get_rtc_time());
}
```

@donghengqaz I suggest declare these system functions in some header because it's confusing and it can help intellisense or auto-complete in IDE to work properly.

thisdream, 2019-11-07:
@Fonger @donghengqaz  thank your for help,I deal with this problem by your help

closed 2019-11-08 by FayeY