espressif/ESP8266_RTOS_SDK #675: 2+MB_flash OTA example not working

berkutta, 2019-08-09:
It looks like the example `2+MB_flash` is not working on the current master:
```
I (8091) ota: Have written image length 285084
I (8091) ota: Have written image length 286108
I (8101) ota: Have written image length 287132
I (8111) ota: Have written image length 288156
I (8121) ota: Have written image length 289180
I (8121) ota: Have written image length 290204
I (8131) ota: Have written image length 291228
I (8141) ota: Have written image length 291796
I (8151) ota: Have written image length 292820
I (8151) ota: Have written image length 293472
I (8161) ota: Connection closed, all packets received
I (8171) ota: Total Write binary data length : 293472
I (8171) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x3a8c0 (239808) map
0x40210010: _flash_cache_start at ??:?

I (8251) esp_image: segment 1: paddr=0x0014a8d8 vaddr=0x4024a8d0 size=0x074b8 ( 29880) map
I (8261) esp_image: segment 2: paddr=0x00151d98 vaddr=0x3ffe8000 size=0x0057c (  1404) 
I (8261) esp_image: segment 3: paddr=0x0015231c vaddr=0x40100000 size=0x008c0 (  2240) 
I (8271) esp_image: segment 4: paddr=0x00152be4 vaddr=0x401008c0 size=0x04e50 ( 20048) 
I (8291) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x3a8c0 (239808) map
0x40210010: _flash_cache_start at ??:?

I (8361) esp_image: segment 1: paddr=0x0014a8d8 vaddr=0x4024a8d0 size=0x074b8 ( 29880) map
I (8371) esp_image: segment 2: paddr=0x00151d98 vaddr=0x3ffe8000 size=0x0057c (  1404) 
I (8381) esp_image: segment 3: paddr=0x0015231c vaddr=0x40100000 size=0x008c0 (  2240) 
I (8391) esp_image: segment 4: paddr=0x00152be4 vaddr=0x401008c0 size=0x04e50 ( 20048) 
I (8421) ota: Prepare to restart system!
E (8421) ota: Disconnect reason : 8

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7204, room 16 
tail 4
chksum 0xca
load 0x3ffe8408, len 24, room 4 
tail 4
chksum 0x43
load 0x3ffe8420, len 3544, room 4 
tail 4
chksum 0x47
I (60) boot: ESP-IDF v3.2-39-g6ada8123 2nd stage bootloader
I (60) boot: compile time 15:58:55
I (69) qio_mode: Enabling default flash chip QIO
I (70) boot: SPI Speed      : 40MHz
I (75) boot: SPI Mode       : QIO
I (81) boot: SPI Flash Size : 2MB
I (87) boot: Partition Table:
I (93) boot: ## Label            Usage          Type ST Offset   Length
I (104) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (115) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (127) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (139) boot:  3 ota_0            OTA app          00 10 00010000 000f0000
I (150) boot:  4 ota_1            OTA app          00 11 00110000 000f0000
I (162) boot: End of partition table
I (168) esp_image: segment 0: paddr=0x00110010 vaddr=0x40210010 size=0x3a8c0 (239808) map
0x40210010: _flash_cache_start at ??:?

I (262) esp_image: segment 1: paddr=0x0014a8d8 vaddr=0x4024a8d0 size=0x074b8 ( 29880) map
I (273) esp_image: segment 2: paddr=0x00151d98 vaddr=0x3ffe8000 size=0x0057c (  1404) load
I (274) esp_image: segment 3: paddr=0x0015231c vaddr=0x40100000 size=0x008c0 (  2240) load
I (287) esp_image: segment 4: paddr=0x00152be4 vaddr=0x401008c0 size=0x04e50 ( 20048) load
I (307) boot: Loaded app from partition at offset 0x110000
Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x402211cb, excvaddr=0x0010fff0, depc=0x00000000
0x402211cb: esp_internal_idle_hook at /home/bm/esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:379

Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x402211cb, excvaddr=0x0010fff0, depc=0x00000000
0x402211cb: esp_internal_idle_hook at /home/bm/esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:379

Fatal exception (28): 
epc1=0x40001800, epc2=0x00000000, epc3=0x402211cb, excvaddr=0x0010fff0, depc=0x00000000
0x402211cb: esp_internal_idle_hook at /home/bm/esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:379
```

donghengqaz, 2019-08-12:
Thanks for your report. It is a new bug when I add "make size" function, and I will fix it as soon as possible.

colesnicov, 2019-08-25:
```
I (32) boot: ESP-IDF v3.2-59-g86808496 2nd stage bootloader
I (32) boot: compile time 15:41:16
I (32) boot: SPI Speed      : 80MHz
I (34) boot: SPI Mode       : DIO
I (37) boot: SPI Flash Size : 4MB
I (40) boot: Partition Table:
I (42) boot: ## Label            Usage          Type ST Offset   Length
I (47) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (51) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (56) boot:  2 factory          factory app      00 00 00010000 000f0000
I (61) boot: End of partition table
I (64) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x45e28 (286248) map
0x40210010: _flash_cache_start at ??:?

I (130) esp_image: segment 1: paddr=0x00055e40 vaddr=0x40255e38 size=0x0d25c ( 53852) map
I (142) esp_image: segment 2: paddr=0x000630a4 vaddr=0x60001200 size=0x00060 (    96) load
I (142) esp_image: segment 3: paddr=0x0006310c vaddr=0x3ffe8000 size=0x00994 (  2452) load
I (147) esp_image: segment 4: paddr=0x00063aa8 vaddr=0x40100000 size=0x06e00 ( 28160) load
I (158) esp_image: segment 5: paddr=0x0006a8b0 vaddr=0x40106e00 size=0x04f0c ( 20236) load
I (163) boot: Loaded app from partition at offset 0x10000

```

closed 2019-08-25 by berkutta