espressif/ESP8266_RTOS_SDK #676: Missing dependency pyparsing

cranphin, 2019-08-11:
See this comment:
https://github.com/espressif/ESP8266_RTOS_SDK/commit/6ada81237e7e75bd1e9628547e360a350e0284ed#r34648275

A dependency on pyparsing was introduced, without adding it to requirements.txt (or documenting it on https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/linux-setup.html).

So now my automated builds fail on:
```
Generating esp8266.project.ld
Traceback (most recent call last):
  File "/esp/ESP8266_RTOS_SDK/tools/ldgen/ldgen.py", line 26, in <module>
    from fragments import FragmentFile
  File "/esp/ESP8266_RTOS_SDK/tools/ldgen/fragments.py", line 19, in <module>
    from sdkconfig import SDKConfig
  File "/esp/ESP8266_RTOS_SDK/tools/ldgen/sdkconfig.py", line 18, in <module>
    from pyparsing import Word, alphanums, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
ImportError: No module named pyparsing
make: *** [/esp/project_template/build/esp8266/esp8266.project.ld] Error 1
/esp/ESP8266_RTOS_SDK/components/esp8266/Makefile.projbuild:72: recipe for target '/esp/project_template/build/esp8266/esp8266.project.ld' failed
The command '/bin/sh -c cp -r $IDF_PATH/examples/get-started/project_template . && cd project_template && make -j8 defconfig && make -j8 all && cd .. && rm -rf project_template' returned a non-zero code: 2
The command "docker build --no-cache -t cranphin/esp8266-idf ." exited with 2.
```

Should be easy to fix :)

donghengqaz, 2019-08-12:
Thanks for report and I will update your mention to requirements.txt.

cranphin, 2019-08-12:
Automated build works again, fixed, thanks :)

closed 2019-08-12 by cranphin