espressif/ESP8266_RTOS_SDK #680: Make time critical sections (microsecond accurate) possible with WiFi on

cranphin, 2019-08-13:
## Problem Description

I'm trying to do 1-wire bit banging, where timing should be near microsecond accurate for gpio write and read.
Since the protocol is robust, accuracy is only needed while writing or reading a bit (15-60 us long critical sections).
But I can't get this to work reliably when WiFi is sending, even when doing all of:
- Changing task priority to 15 (highest)
- Disabling -all- interrupts using XTOS_DISABLE_ALL_INTERRUPTS
- Adding IRAM_ATTR

With all of this, sometimes I still get a 10us extra delay in my timing/critical code.
Is there anything else I can disable/change? Any idea what might still be interrupting my code? :)

Note that WiFi is still stable like this, and my critical sections are very short, so I hope it should be possible :)

See also: https://bbs.espressif.com/viewtopic.php?f=7&t=9553

### Code to reproduce this issue

Task which runs the code:
```cpp
	const BaseType_t ret = xTaskCreate(ds_task,
					"ds_task",
					2048,
					NULL,
					// See https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/api-guides/system-tasks.html for prio's
					15, // Can't go higher, who cares about stable WiFi.
					&ds_task_handle);
```

Includes/defines
```cpp
#include <xtensa/xtruntime.h>
#include "esp8266/gpio_struct.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

// At microsecond speeds, the functions from gpio.h are too heavy
#define GPIO_FAST_SET_1(gpio_num) GPIO.out_w1ts |= (0x1 << gpio_num)
#define GPIO_FAST_SET_0(gpio_num) GPIO.out_w1tc |= (0x1 << gpio_num)
#define GPIO_FAST_OUTPUT_ENABLE(gpio_num) GPIO.enable_w1ts |= (0x1 << gpio_num)
#define GPIO_FAST_OUTPUT_DISABLE(gpio_num) GPIO.enable_w1tc |= (0x1 << gpio_num)
#define GPIO_FAST_GET_LEVEL(gpio_num) ((GPIO.in >> gpio_num) & 0x1)
```

The method with the critical section
```cpp
uint32_t IRAM_ATTR onewire_read_bit(void) {
	uint32_t r;

	uint32_t savedLevel = XTOS_DISABLE_ALL_INTERRUPTS;

	GPIO_FAST_OUTPUT_ENABLE(ONEWIRE_PIN);
	GPIO_FAST_SET_0(ONEWIRE_PIN);

	ets_delay_us(3);

	GPIO_FAST_OUTPUT_DISABLE(ONEWIRE_PIN);

	ets_delay_us(10); // Somewhere near here, sometimes we still get 10us extra delay

	r = GPIO_FAST_GET_LEVEL(ONEWIRE_PIN);

	XTOS_RESTORE_INTLEVEL(savedLevel);

	usleep(53);

	return r;
}```

dsptech, 2019-08-17:
Hi,
I never seen a such case, but if you get an extra latency inside the "critical section" (interrupts totally disabled, even NMI), I suppose that ets_delay_us() perform other jobs other than delays. 
I think that you could try to write your own micro_us delay function by using the soc_get_ccount() macro. 
The macro return the current cpu clock tick number.
Regards.

cranphin, 2019-08-17:
I've always assumed ets_delay_us was not much more than a loop.
My assembler is a bit rusty, but someone has got the code for ets_delay_us  :)
https://github.com/trebisky/esp8266/blob/a886c2ff582d9c368f5d9d669f31cc66105a44ad/reverse/bootrom/boot_OLD.txt#L5510
https://github.com/trebisky/esp8266/blob/a886c2ff582d9c368f5d9d669f31cc66105a44ad/reverse/bootrom/boot.txt#L6117
At a glance I don't see anything special going on

Maybe I didn't actually turn of all the interrupts somehow, or I'm missing something else. I'll try to make a smaller sample focused on just this :) I'm using a logic analyzer to get the timings, so I'm pretty certain about the delay, things are as expected for many bits sent, and then there's a sudden delay. Small, but enough to miss the pulse I need to check :)

Note to self: look at this too https://sub.nanona.fi/esp8266/timing-and-ticks.html

cranphin, 2019-08-17:
I've copied pasted the ets_delay_us assmbly and comments from the links from my previous comment:
```
ets_delay_us:
40002ecc:   12c1f0      addi    a1, a1, -16           ; stack
40002ecf:   d26102      s32i    a13, a1, 8            ; save a13
40002ed2:   c911        s32i.n  a12, a1, 4            ; save a12
40002ed4:   0901        s32i.n  a0, a1, 0             ; save return

40002ed6:   cd02        mov.n   a12, a2
40002ed8:   c5e50a      call0   xthal_get_ccount      ; 0x4000dd38
                      
40002edb:   01fbff      l32r    a0, [0x3fffc704]      ; 0x40002ec8
40002ede:   0800        l32i.n  a0, a0, 0             ; reads 80 if 80 Mhz
40002ee0:   dd02        mov.n   a13, a2
40002ee2:   c0c082      mull    a12, a0, a12
40002ee5:   05e50a      call0   xthal_get_ccount      ; 0x4000dd38
40002ee8:   d032c0      sub a3, a2, a13
40002eeb:   c7b308      bgeu    a3, a12, 0x40002ef7

; polling loop does delay
40002eee:   85e40a      call0   xthal_get_ccount      ; 0x4000dd38
40002ef1:   d042c0      sub a4, a2, a13
40002ef4:   c734f6      bltu    a4, a12, 0x40002eee
40002ef7:   c811        l32i.n  a12, a1, 4            ; restore a12
40002ef9:   d821        l32i.n  a13, a1, 8            ; restore a13
40002efb:   0801        l32i.n  a0, a1, 0             ; return
40002efd:   12c110      addi    a1, a1, 16            ; stack
40002f00:   0df0        ret.n

40002f02:   0000
```

dsptech, 2019-08-17:
Hi,
the assembler is rusty also to me, but I agree: this code is a simple delay loop. 
Regarding interrupts disabling, I think that there is no other ways.

Sorry, I have no other realistic ideas. I think that the question need a reply by espressif staff.
Regards.

cranphin, 2019-08-18:
Created a fairly minimal sample to display the issue:
https://github.com/cranphin/TimingTest/tree/6701850173ccb78df91525c44b5091bea9eae0b1

It does some delay + gpio, and measures the timing using ccount.
Most of the time, the measurement shows 14us (1170 cpu cycles at 80mhz).
But with WiFi sending, now and then there's a longer delay, which it logs.
Example run (the last value is delay in usec, which should be 14, but it's more):
I (4895) main: 7:2304(28)
I (7472) main: 35:2221(27)
I (12418) main: 1:1651(20)
I (12935) main: 24:2111(26)
I (14379) main: 26:1408(17)
I (14382) main: 36:1935(24)
I (19431) main: 5:2264(28)
I (26954) main: 27:1759(21)
I (30973) main: 8:2679(33)
I (36125) main: 2:2262(28)
I (40660) main: 39:2165(27)
I (40765) main: 46:2037(25)
I (42414) main: 9:2154(26)
I (44888) main: 29:1548(19)
I (48392) main: 14:1999(24)
I (50660) main: 39:1488(18)
I (52413) main: 34:1598(19)

I really want to know what causes the delay in this sample, and how to turn it off :)

cranphin, 2019-08-18:
Even smaller example, removed gpio:
https://github.com/cranphin/TimingTest/tree/2c18f12c6d1e036ffbda2f7b5fec3b476c403781

cranphin, 2019-08-18:
And semi-interestingly, even removing MQTT I still see some delays, but less often. But I guess WiFi still does things, even without TCP traffic :)

donghengqaz, 2019-08-20:
To realize the real time controlling, you should do following:

1. close all hardware exception by API 
`vPortETSIntrLock`(exit one is `vPortETSIntrUnlock`)
2. the API `ets_delay_us` may not cause the real time delaying. because the previous coding boy had not considered the time cost of the API it self, so you may reduce some the inputting time count value by testing

I hope the upper advices can help you.

cranphin, 2019-08-20:
> I hope the upper advices can help you.

First test shows vPortETSIntrLock() and vPortETSIntrUnlock (with #include "rom/ets_sys.h") improving timing stability a lot! :)
I still get a spurious 2us delay (179 ticks), but that's small enough to not care about :)
And have to investigate more still! :)

It looks like these mystical lines are the special sauce (on top of disabling interrupts):
Open
```c++
            do {
                REG_WRITE(INT_ENA_WDEV, WDEV_TSF0_REACH_INT);
            } while(REG_READ(INT_ENA_WDEV) != WDEV_TSF0_REACH_INT);
```

Close
```c++
            extern uint32_t WDEV_INTEREST_EVENT;
            REG_WRITE(INT_ENA_WDEV, WDEV_INTEREST_EVENT);
```

Seems wdev is WiFi related, and not well documented or understood, but I'm happy as long as this works ^_^

dsptech, 2019-08-20:
Very precious information here !
Thank you @donghengqaz

cranphin, 2019-08-20:
And another bit, the occasional 2us delay I still saw disappeared by setting my task priority to 15.
So maybe some task switching was still happening :)

So with that, here's an updated sample which has very predictable timing.
Only the first fake 'read' in my loop still has varied timing, but I think it's just how the compiler/cpu deals with things, caching and such, not being interrupted. And the change is small, with this it should be easy to make a reliable 1-wire bus :)
https://github.com/cranphin/TimingTest/tree/58f0e33fc50a7613fd2872162ba5409580d86e97

AntoinedeChassey, 2019-11-26:
Thanks for sharing @cranphin, I found your issue while having some random errors of DS18B20 temperature sensor readings.
Sometimes, I randomly get bad measurements, probably because of this us delay from the WiFi stack (and because my sensor reading task priority was set to 0).
I will try with setting it to 15 and see how it behaves!
Thanks!

namezys, 2020-03-22:
I got similar problem.
I'm trying to write very precisely delay function which based on cycle count.

```
inline
void __attribute__ ((always_inline)) Wait(uint32_t delta)
{
    uint32_t cycleCount;
    uint32_t waitUntil;
    __asm__ __volatile__("rsr     %0, ccount":"=a" (waitUntil));
    waitUntil += delta;
    do {
        __asm__ __volatile__("rsr     %0, ccount":"=a" (cycleCount));
    } while(waitUntil > cycleCount);
}
```
which will be compiled in pretty good asm code (example because it's inline function):
```
        # a2 allocated for waitUntil, a4 for cycleCount, delta is 9
	rsr     a2, ccount	# waitUntil
	addi.n	a2, a2, 9	# waitUntil, waitUntil,
LOOP:
	rsr     a4, ccount	# cycleCount
	bltu	a4, a2, LOOP	# cycleCount, waitUntil,
```
it's very predictable and contains only 4 instruction. Oblivious, it can't be 1-cycle precise but I can predict it's graduation.

Next I created a set of function with name `testWait##n` where is n a constant.
For example, `textWait7`
```
uint32_t __attribute__((noinline))
testWait7()
 {
        uint32_t start;
        __asm__ __volatile__("rsr     %0, ccount":"=a" (start));
        Wait(7);
        uint32_t end;
        __asm__ __volatile__("rsr     %0, ccount":"=a" (end));
        return end - start - 1;
 }
```
I want to measure how much time it spend inside Wait.
Asm again is pretty beautiful and don't use stack at all
```_Z9testWait7v:
        # a3 - start
        # a2 - end (is used after Wait finished)
        # From wait:
        #   a2 - waitUntil
        #   a4 - cycleCount
	rsr     a3, ccount	# start
	rsr     a2, ccount	# waitUntil
	addi.n	a2, a2, 7	# waitUntil, waitUntil,
.LOOP:
	rsr     a4, ccount	# cycleCount
	bltu	a4, a2, LOOP	# cycleCount, waitUntil,

	rsr     a2, ccount	# end
	movi.n	a4, -1		# next 3 instruction do end - start - 1 in some magic way
	xor	a3, a4, a3	# 
	add.n	a2, a3, a2	# 
	ret.n
```
Looks very predictable.
Same asm code I got for testWait0 ... testWait9

Next step is running their in row inside critical section.
```
void testWait(uint32_t constDelays[10])
{
    ENTER_SOME_CRITICAL_SECTION;
    constDelays[0] = testWait0();
    constDelays[1] = testWait1();
    ....
    constDelays[9] = testWait2();
    EXIT_SOME_CRITICAL_SECTION
}
```
Asm looks good (for example)
```
	addi	sp, sp, -16	#,,
	s32i.n	a0, sp, 12	#,
	s32i.n	a12, sp, 8	#,
	mov.n	a12, a2	# constDelays, constDelays
	call0	vPortETSIntrLock	#
	call0	_Z9testWait9v	#
	s32i.n	a2, a12, 36	# MEM[(uint32_t *)constDelays_3(D) + 36B],
	call0	_Z9testWait0v	#
	s32i.n	a2, a12, 0	# *constDelays_3(D),
	call0	_Z9testWait1v	#
	s32i.n	a2, a12, 4	# MEM[(uint32_t *)constDelays_3(D) + 4B],
	call0	_Z9testWait2v	#
	s32i.n	a2, a12, 8	# MEM[(uint32_t *)constDelays_3(D) + 8B],
	call0	_Z9testWait3v	#
	s32i.n	a2, a12, 12	# MEM[(uint32_t *)constDelays_3(D) + 12B],
	call0	_Z9testWait4v	#
	s32i.n	a2, a12, 16	# MEM[(uint32_t *)constDelays_3(D) + 16B],
	call0	_Z9testWait5v	#
	s32i.n	a2, a12, 20	# MEM[(uint32_t *)constDelays_3(D) + 20B],
	call0	_Z9testWait6v	#
	s32i.n	a2, a12, 24	# MEM[(uint32_t *)constDelays_3(D) + 24B],
	call0	_Z9testWait7v	#
	s32i.n	a2, a12, 28	# MEM[(uint32_t *)constDelays_3(D) + 28B],
	call0	_Z9testWait8v	#
	s32i.n	a2, a12, 32	# MEM[(uint32_t *)constDelays_3(D) + 32B],
	call0	vPortETSIntrUnlock	#
	l32i.n	a0, sp, 12	#,
	l32i.n	a12, sp, 8	#,
	addi	sp, sp, 16	#,,
	ret.n
```

I expected that inside critical section nobody can interrupt me and I would get predicted result. But
```
I (1292) TEST: run 0 time 4
I (1294) TEST: run 1 time 6
I (1295) TEST: run 2 time 6
I (1297) TEST: run 3 time 183
I (1299) TEST: run 4 time 185
I (1304) TEST: run 5 time 185
I (1309) TEST: run 6 time 190
I (1315) TEST: run 7 time 11
I (1321) TEST: run 8 time 11
I (1326) TEST: run 9 time 16
```

Output is doing after finish `testWait` function so it can't affect it.
Changing order of call inside `testWait` can change place of fluctuation.
I think that this fluctuation is about 179 or 177 cycles.

I tried:
* standard enter_CRITICAL_SECTION
* vPortETSIntrLock
* uint32_t savedLevel = XTOS_DISABLE_ALL_INTERRUPTS;

But results were same.

I'm sure that somebody interrupt me but I don't know how to avoid it.
Also I know that length of critical section in a real project can be up to 4ms and hope that it's ok.

namezys, 2020-03-22:
After investigation: Looks like the CPU spend time to load code from flash.
I appended attribute `IRAM_ATTR` to `testWait` functions and it's work pretty good now.