espressif/ESP8266_RTOS_SDK #690: MQTT with SSL error in connecting

yuguitao, 2019-09-06:
error log as below:

> [{"clientId":"DEV_1144513846","ip":"33.22.2.153","port":1883,"topics":[{"indentifier":"service","isSubPub":1,"level":1,"name":"service/null/null"},{"indentifier":"control","isSubPub":1,"level":1,"name":"control/null/null"},{"indentifier":"sync","isSubPub":2,"level":1,"name":"sync/null/null"},{"indentifier":"CustomEvent","isSubPub":2,"level":1,"name":"CustomEvent/null"}],"vendorId":"xxxx"}]
I (4815) xxxx_http: start to call http user routine
I (4824) xxxx_http: id = DEV_1144513846
I (4826) xxxx_http: ip = 33.22.2.153
I (4830) xxxx_http: port = 1883
I (4834) xxxx_http: service topic = service/e6add2a573234a9691fe9e6b9297e092/1144513846
I (4843) xxxx_http: control topic = control/e6add2a573234a9691fe9e6b9297e092/1144513846
I (4852) xxxx_http: sync topic = sync/e6add2a573234a9691fe9e6b9297e092/1144513846
I (4860) xxxx_http: reply topic = reply/e6add2a573234a9691fe9e6b9297e092/1144513846
I (4951) dev_config: Sended uart data len = 0xc.
Sended uart data = f4 f5 00 08 01 00 01 01 00 00 00 08 
W (5871) xxxx_mqtt: ###mqtt client thread start
E (5875) xxxx_mqtt: Broker_URL = mqtts://33.22.2.153:1883
I (5879) xxxx_mqtt: [APP] Free memory: 20392 bytes
D (5882) MQTT_CLIENT: MQTT client_id=DEV_1144513846
D (5886) MQTT_CLIENT: Core selection enabled on 0
D (5881) esp-tls: host:33.22.2.153: strlen 11
**E (5895) esp-tls: couldn't get hostname for :33.22.2.153:**
E (5901) esp-tls: Failed to open new connection
E (5906) TRANS_SSL: Failed to open a new connection
E (5911) MQTT_CLIENT: Error transport connect
I (5916) MQTT_CLIENT: Reconnect after 10000 ms

yuguitao, 2019-09-06:
hardware: 8266 
sdk: 3.2

yuguitao, 2019-09-07:
fixed

closed 2019-09-07 by yuguitao

sky9971, 2019-11-21:
Hi @yuguitao ,
I am also facing the same issue while using the ESP MQTT client.
Can you please tell me how did you solve this?