espressif/ESP8266_RTOS_SDK #693: os_timer 和 esp_timer  在 master branch 下沒作用了 (release/3.2正常)

Fonger, 2019-09-09:
## Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    cafe4468
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description

`os_timer_setfn` `os_timer_setarm`
no longer  work in master branch. It works in release/3.2 branch.
在 master 下沒作用了，但在 release/3.2 branch 下正常

### Expected Behavior
It should work.

### Actual Behavior
It compiles but it doesn't work.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

static void timer_func(void *arg) {
  printf("hello world!\n");
}
os_timer_t timer;
void app_main()
{
        os_timer_setfn(&timer, timer_func, NULL);
        os_timer_arm(&timer, 1000, 1);
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

esp_timer doesn't work too.
```c
#include "esp_timer.h"
#include "ets_sys.h"
static void timer_func(void *arg)
{
    printf("hello world!\n");
}
esp_timer_handle_t timer;
esp_timer_create_args_t timer_args = {.callback = timer_func, .arg = NULL, .dispatch_method = ESP_TIMER_TASK, .name = "test timer"};
void app_main()
{
    esp_timer_create(&timer_args, &timer);
    esp_timer_start_periodic(timer, 1000);
}
```

另外想問一下以下問題：

# 1. os_timer esp_timer 有何不同
`os_timer` 和 `esp_timer` 有什麼不同？這兩個應該都是軟件時鐘吧？

# 2. FRC2 definition
esp8266應該有兩個hardware timer: frc1, frc2
hw_timer 使用了 frc1 不過因為call overhead 太高
我透過 timer_struct.h 的 frc1 直接去操作 timer
不過上面還沒有 frc2 struct 的定義 但 timer_register.h 有

我目前已經有用到 frc1 需要高精度計時用 timer_re
但 frc2 似乎還沒有對應的接口？

# PWM MacTimer FRC1 FRC2
pwm 中code用到的 MacTimer 又是什麼？我在用 frc1 時會和 pwm 產生干擾嗎
這種情況下要用到第二個硬件時鐘是不是需要 frc2 來解決?

# RTOS Timer
FreeRTOS 的 xTimerCreate 又是用何種方式實現的呢？在 RTOS 下，軟體時鐘、毫秒等級的 timer 是否用 RTOS 的方法較好？

# 獲取時間的方式
esp_get_time() us precision: uint32 這個超時會歸零嗎?
esp_timer_get_time() ms precision: int64

esp_get_time() 沒有定義 要自己定義
是否可以將此定義放在某個 header 文件內 讓 auto completion works?

Fonger, 2019-09-12:
Edit: it works now but I have the same questions.

closed 2019-11-12 by Fonger

zheyuejiang, 2020-08-25:
最近在尝试RTOS V3.3，发现系统时间不准，使用的是esp_timer_get_time().现在想使用esp_get_time()，但是无法找到这个函数，请问怎么使用？