espressif/ESP8266_RTOS_SDK #695: Smartconfig does not provide a method to get airkiss random.

KNpTrue, 2019-09-11:
## Environment

- Development Kit:
- IDF version: master 
- Development Env:     Make
- Operating System:     MacOS
- Power Supply:         USB

## Problem Description

  Because the project needs to get airkiss random to compose the broadcast package, but smartconfig does not provide a method to get airkiss random.

ESP32 esp-idf allows us to get random in the event, like this:
```
} else if (event_base == SC_EVENT && event_id == SC_EVENT_GOT_SSID_PSWD) {
        aks_log(LOG_INFO, "Got SSID and password");

        smartconfig_event_got_ssid_pswd_t *evt = (smartconfig_event_got_ssid_pswd_t *)event_data;
	struct al_aks_data *data = al_os_mem_calloc(sizeof(struct al_aks_data));
	if (!data) {
		aks_log(LOG_ERR, "alloc data error.");
		return;
	}
        memcpy(data->ssid.id, evt->ssid, sizeof(evt->ssid));
        memcpy(data->password, evt->password, sizeof(evt->password));
	data->ssid.len = strlen((char *)evt->ssid);
extern u8 aks_random;
	aks_random = evt->token;
```
I would be grateful if esp8266 could provide an interface to get random as soon as possible.

xuhongv, 2019-10-21:
- @KNpTrue 参考这个，支持近场发现自定义消息 https://github.com/xuhongv/xLibEsp8266Rtos3.1AirKiss

KNpTrue, 2019-10-25:
@xuhongv 我知道可以调用airkiss 的库，我之后的做法也是直接调用，只是说esp8266 中的smartconfig缺失这个有点遗憾

niurenyige, 2020-05-09:
这个有更新吗？

KNpTrue, 2020-05-19:
> 这个有更新吗？

不清楚，你可以找找airkiss的开源库