espressif/ESP8266_RTOS_SDK #702: unreasonable crash

iravan, 2019-09-18:
```
Task stack [tiT] stack from [0x3fff2bfc] to [0x3fff3bf8], total [4096] size

                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c

3fff3ac0  0x3ffefa5c 0x401003e1 0x00000030 0x4010042c 0x3fff3b20 0x00000000 0x3ffefa5c 0x40234649 0x3fff1ab8 0x3fff2bac 0xffffffff 0x00000003 0x00000000 0x00000000 0x00000070 0x3ffee2e4
0x401003e1: vPortExitCritical at /esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:368

0x4010042c: PendSV at /esp/ESP8266_RTOS_SDK/components/freertos/port/esp8266/port.c:368

0x40234649: xQueueGenericSend at /esp/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332

3fff3b00  0xffffffff 0x00000000 0x3fff3bd0 0x00000000 0x00000000 0x3ffef834 0x3ffefa5c 0x4023a461 0x00000000 0x00000001 0x3fff2b80 0x4023493a 0x00000000 0x00001ca2 0xfffffffe 0x40240085
0x4023a461: sys_mbox_trypost at /esp/ESP8266_RTOS_SDK/components/lwip/port/esp8266/freertos/sys_arch.c:398

0x4023493a: xQueueReceive at /esp/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332

0x40240085: recv_tcp at /esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/api/api_msg.c:293

3fff3b40  0x40105e20 0x00000013 0x3ffefad4 0x40105e48 0x3ffefb34 0x40105e2c 0x40105e2d 0x4023c819 0x29cc0a55 0x00000011 0x00000000 0x4023a4ae 0x00000000 0x00000000 0x3ffefdbc 0x3ffefcba
0x40105e20: xthal_set_intset at ??:?

0x40105e48: xthal_set_intset at ??:?

0x40105e2c: xthal_set_intset at ??:?

0x40105e2d: xthal_set_intset at ??:?

0x4023c819: tcp_input at /esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/tcp_in.c:476 (discriminator 1)

0x4023a4ae: sys_arch_mbox_fetch at /esp/ESP8266_RTOS_SDK/components/lwip/port/esp8266/freertos/sys_arch.c:398

3fff3b80  0x40106e14 0x3ffefad4 0x40105e10 0x40239501 0x40106e14 0x00000014 0x00000001 0x4023613c 0x40105de4 0x3fff3bd0 0x3fff6940 0x00000000 0x40106e14 0x00000008 0x3ffefad4 0x40239d7d
0x40105e10: xthal_set_intset at ??:?

0x40239501: ip4_input at /esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/ipv4/ip4.c:685

0x4023613c: sys_timeouts_mbox_fetch at /esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/timeouts.c:427

0x40105de4: xthal_set_intset at ??:?

0x40239d7d: ethernet_input at /esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/netif/ethernet.c:182

3fff3bc0  0x00000000 0x00000000 0x3fff49c0 0x4023a6b9 0x3ffefaf8 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x4023a6b9: tcpip_thread at /esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/api/tcpip.c:459

        PC: 0x402343eb        PS: 0x00000033        A0: 0x40234656        A1: 0x3fff3ac0
0x402343eb: prvNotifyQueueSetContainer at /esp/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332

0x40234656: xQueueGenericSend at /esp/ESP8266_RTOS_SDK/components/freertos/freertos/queue.c:2332

        A2: 0x3ffefa5c        A3: 0x00000000        A4: 0x96ae041a        A5: 0xe8770f26
        A6: 0x0000001c        A7: 0x000000a2        A8: 0x23493a3f        A9: 0x00000040
       A10: 0x2434c320       A11: 0x00000003       A12: 0x3ffefa5c       A13: 0x2c1e51cb
       A14: 0x3fff3b20       A15: 0xfffffffe       SAR: 0x00000018  EXCCAUSE: 0x00000009

Task stack [tiT] stack from [0x3fff2bfc] to [0x3fff3bf8], total [4096] size

register map is 3ffee500 error
                   0          4          8          c         10         14         18         1c         20         24         28         2c         30         34         38         3c

3fff2bfc  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fff2c3c  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fff2c7c  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fff2cbc  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fff2cfc  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fff2d3c  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fff2d7c  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fff2dbc  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fff2dfc  0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[... 193 lines cut ...]

0x40239501: ip4_input at /esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/ipv4/ip4.c:685
0x4023613c: sys_timeouts_mbox_fetch at /esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/core/timeouts.c:427

0x40105de4: xthal_set_intset at ??:?

3fff3bbc  0x40239d7d 0x00000000 0x00000000 0x3fff49c0 0x4023a6b9 0x3ffefaf8 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x40239d7d: ethernet_input at /esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/netif/ethernet.c:182

0x4023a6b9: tcpip_thread at /esp/ESP8266_RTOS_SDK/components/lwip/lwip/src/api/tcpip.c:459

        PC: 0x4010610a        PS: 0x00000033        A0: 0x4010610a        A1: 0x3ffee500
0x4010610a: xthal_set_intset at ??:?

0x4010610a: xthal_set_intset at ??:?

        A2: 0x3ffe9e6c        A3: 0x3ffea300        A4: 0x00000000        A5: 0x3ffea31c
        A6: 0x3ffea310        A7: 0x00000001        A8: 0x00000000        A9: 0x401061c8
0x401061c8: xthal_set_intset at ??:?

       A10: 0x00000000       A11: 0x00000075       A12: 0x00000003       A13: 0x402229d3
0x402229d3: esp_wifi_disconnect at ??:?

rlr#n  A14: 0x0000001d       A15: 0x40105380       SAR: 0x0000001d  EXCCAUSE: 0x00000000
```

ustccw, 2019-09-18:
@iravan would you mind providing all the test code and steps for us to reproduce it easily?

27-301, 2019-09-19:
@iravan Hope you can try current master branch, now sdk can support backtrace, which is a useful facility to the debug.

closed 2019-11-08 by FayeY