espressif/ESP8266_RTOS_SDK #703: lwip_connect sometimes has endless blocking

iravan, 2019-09-18:
`lwip_connect` sometimes just hang on the function itself without timeout.

ustccw, 2019-09-18:
@iravan Please try to use `connect` instead of `lwip_connect`, same to all `lwip_xxx` APIs.
Moreover, please try to set timeout by `setsockopt` if endless blocking.
good luck.

iravan, 2019-09-18:
Yeah , I see. I just specify the lib I used. 
As I know the `setsockopt` can only config the tcp `send` or `recv` API by using flag `SO_RCVTIMEO` and `SO_SNDTIMEO`. The `setsockopt` will not affect `connect`. 
I know `connect` API can be non-blocking function. But it mustn't be endless blocking. It has to have a reasonable timeout.

ustccw, 2019-09-27:
@iravan maybe the following code would be a little help for you.
```
// set socket to non-blocking
int ret = connect(fd, ...);

if (ret != 0) {
    if (errno == EINPROGRESS) {
        // connecting
        ret = select(fd + 1, &readset, &writeset, NULL, &timeout);

        if (ret < 0) {
        // failed
        } else if (ret == 0) {
        // timeout
        } else {
            if (ret == 1 && FD_ISSET(fd, &writeset)) { 
                // connect OK 
            } else {
                // connect error
            }
        }
    } else {
    // connect failed
    }
}
```

closed 2019-11-08 by FayeY