espressif/ESP8266_RTOS_SDK #705: FOTA OLD TO NEW FAILS AT REBOOT

Tonix22, 2019-09-18:
##Environment
SDK1.5-->SDK3.2 
ESP8266 4MB 
Model esp12f

SDK1.5 -->512KB, 512KB
SDK3.2 --> partition table

I have SPIFFS in both of them. 
nvs,           data,    nvs,      0x9000,  0x4000,
otadata,   data,     ota,      0xD000,  0x2000,
phy_init,  data,     phy,       0xf000,  0x1000,
ota_0,           0,   ota_0,    0x01000,  0x7D000,
ota_1,           0,    ota_1,   0x81000, 0x7D000,
storage,   data, spiffs,  ,        448K, 

##Problem Description

In SDK1.5 I download the OTA compressed file make by the sdk3.2  [Example.v2_to_v3.ota.bin]
When the download finishes and system reboot the ESP hangs in Fatal exception (29):
The download was in a custom lwip socket, I just download data by chunks and save them in the other partition, this step is quite good because I can do FOTA from 1.5 to 1.5 without  issue. The problem is from sdk1.5 to sdk3.2

##Expected Behavior
SDK3.2 unpacks the boot loader and make FOTA process transition. 
##Steps to reproduce
FOTA old to new. 
##Debug Logs
2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user2 @ 101000

Fatal exception (29):
epc1=0x4000df23, epc2=0x00000000, epc3=0x00000000, excvaddr=0x1021e211, depc=0x00000000
Fatal exception (29):
epc1=0x4000df23, epc2=0x00000000, epc3=0x00000000, excvaddr=0x1021e211, depc=0x00000000
.......

27-301, 2019-09-19:
You mentioned that you used 512KB + 512KB map in old sdk, but from you log, your old bootloader jump to run user2 @ 101000, which means 1024KB + 1024KB map. You should download new_ota.bin to ota0 or ota1 in 0x1000 or 0x81000. Please make sure your steps firstly. Thanks.

Tonix22, 2019-09-19:
It was the same problem. 

load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user2 @ 81000

Fatal exception (29):
epc1=0x4000df23, epc2=0x00000000, epc3=0x00000000, excvaddr=0x1021e211, depc=0x00000000

27-301, 2019-09-20:
Which example do you use? You need to use examples/system/ota/native_ota/1MB_flash/new_to_new_with_old. Seems that the compressed ota file is not correct. Please dump the context in your flash, first 1MB is OK, and share here. Thanks

berkutta, 2019-09-22:
Same issue for me when trying to upgrade from old SDK `4=4096KB( 512KB+ 512KB)` to new SDK `1MB_flash/new_to_new_with_old`

donghengqaz, 2019-10-14:
From your posting log, I cannot see the "new" bootloader starts and output its log, so I think the booltoader is not filled in the starting address of "Example.v2_to_v3.ota.bin".

So when your OTA fails and it show logs like following:
```
Fatal exception (29):
epc1=0x4000df23, epc2=0x00000000, epc3=0x00000000, excvaddr=0x1021e211, depc=0x00000000
```
You can try download the "Example.v2_to_v3.ota.bin" to the address of "user2" directly and check what next happens.

Otherwise you may supply your "Example.v2_to_v3.ota.bin" to me, I will check if I can do the OTA to upgrade to this firmware.

FayeY, 2019-11-08:
Closing due to lack of response, please feel free to reopen if this is still an issue. Thanks.

closed 2019-11-08 by FayeY