espressif/ESP8266_RTOS_SDK #706: esp_err_to_name.c:406:31: error: unknown type name 'esp_err_t'

jairusmingua, 2019-09-19:


27-301, 2019-09-19:
Can you share some sample code to reproduce this issue?

suryaa-teja, 2019-09-24:
I too was facing the same issue, just comment out these lines in esp_err_to_name.c
`//#if __has_include("esp_err.h")
#include "esp_err.h"
//#endif`

this worked for me.

Surya

donghengqaz, 2019-10-14:
You should use the newest "gcc", and it supports this compiling option. It can be found at README.md at the root directry.

closed 2019-11-08 by FayeY

vortex314, 2020-05-27:
Used latest version of *gcc tensa-lx106-elf-linux64-1.22.0-100-ge567ec7-5.2.0.tar.gz*
and latest version from ESP8266_RTOS_SDK , still same issue. 

CC build/esp_common/src/esp_err_to_name.o
/home/lieven/esp/ESP8266_RTOS_SDK/components/esp_common/src/esp_err_to_name.c:391:29: error: unknown type name 'esp_err_t'
 const char *esp_err_to_name(esp_err_t code)
                             ^
/home/lieven/esp/ESP8266_RTOS_SDK/components/esp_common/src/esp_err_to_name.c:406:31: error: unknown type name 'esp_err_t'
 const char *esp_err_to_name_r(esp_err_t code, char *buf, size_t buflen)
                               ^
cc1: warning: unrecognized command line option "-Wno-frame-address" [enabled by default]
make[1]: *** [/home/lieven/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:292: src/esp_err_to_name.o] Error 1
make: *** [/home/lieven/esp/ESP8266_RTOS_SDK/make/project.mk:571: component-esp_common-build] Error 2
The terminal process terminated with exit code: 2