espressif/ESP8266_RTOS_SDK #707: With the latest release/v3.1 (15aff6d) getting ***ERROR*** A stack overflow in task IDLE has been detected.

fmuller-pi, 2019-09-19:
I keep getting the error below ever since I upgraded from `commit 913a06a9` to `commit 15aff6d` on the branch `release/v3.1`:

`***ERROR*** A stack overflow in task IDLE has been detected.

Core 0 was running in ISR context:

          00000000 00000004 00000008 0000000c 00000010 00000014 00000018 0000001c 00000020 00000024 00000028 0000002c 00000030 00000034 00000038 0000003c 

3ffe82c0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 00000000 60009830 00000000 00000000 00000000 00000000 00000000 ffffffff 

3ffe8300  00000004 3ffedbb4 3fff4008 40250ef0 00000000 00000000 00000000 40250cf2 00000000 00000000 00000000 00000000 00000000 00000000 00000000 4022592c 

3ffe8340  40225944 3ffedbb4 00000001 40250e20 00000000 00000000 00000000 402452c8 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000001 

3ffe8380  00000000 60009830 00000000 00000000 00000000 00000000 00000000 ffffffff 00000004 3ffedbb4 3fff4008 40250ef0 00000000 00000000 00000000 40250cf2 

3ffe83c0  00000000 00000000 3fff3600 4025dc1b 00000000 00000000 00000000 4022592c 40225944 00000080 00000001 40250e20 00002073 3fff35fc 00000000 402452c8 

3ffe8400  3ffe8440 3ffe8430 00000008 3fff19c4 00000004 00000000 3fff22f4 00000001 00000000 60009830 00000004 40107350 40224b34 3fff35fc a5a5a5a5 00000001 

3ffe8440  00000080 00000080 3fff22f4 40250ef0 3ffe9542 00000000 3ffe9540 40245212 00000400 00000400 3ffe85a8 401008b8 00000004 00000000 00000001 40105a98 

3ffe8480  40100a82 00000001 08000000 40100a6c 402451c2 40232c00 3fff230c 402451d5 3fff3258 00000004 3ffe84c4 00000000 00000000 00000000 40105e04 40105e4d 

      PC: 0x402452b5       PS: 0x00000031       A0: 0x3ff7ffe0       A1: 0x3ff7ffe0 

      A2: 0x00000001       A3: 0x40225944       A4: 0x4022592c       A5: 0x00000001 

      A6: 0x00000000       A7: 0x60009830       A8: 0x402107d0       A9: 0x000000f3 

     A10: 0x0000001e      A11: 0x000001e0      A12: 0x3fff4008      A13: 0x3ffedbb4 

     A14: 0x00000004      A15: 0xffffffff      SAR: 0x00000019 EXCCAUSE: 0x00000003`

27-301, 2019-09-20:
Could you please help test with commit 39894e14? so we can check whether it's introduced in commit 15aff6d0. Thanks

fmuller-pi, 2019-09-20:
I just tried with commit `39894e1` and I get the exact same error so that must be the commit that is rising this issue...

donghengqaz, 2019-10-10:
I had told the "phy" developer, and they supplied me a new one, then I tried the new libphy.a, and I could not catch the same crash, so could you please try this ?
[libphy.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3710191/libphy.zip)

FayeY, 2019-11-08:
Is there any update about this issue? Does the libphy above solve the issue?