espressif/ESP8266_RTOS_SDK #708: 做httpserver遇到问题

leovs, 2019-09-20:
esp_err_t get_handler(httpd_req_t *req) {
        char*  buf;
        size_t buf_len = httpd_req_get_url_query_len(req) + 1;
        char* resp_str = (char*) "is ok..";
        httpd_resp_send(req, resp_str, strlen(resp_str));
        return ESP_OK;
}

httpd_uri_t main_uri = {
        .uri       = "/",
        .method    = HTTP_GET,
        .handler   = get_handler
};

httpd_handle_t start_webserver(void) {
        httpd_handle_t server = NULL;
        httpd_config_t config = HTTPD_DEFAULT_CONFIG();

        // Start the httpd server
        ESP_LOGI(TAG, "Starting server on port: '%d'", config.server_port);
        if (httpd_start(&server, &config) == ESP_OK) {
                ESP_LOGI(TAG, "Registering URI handlers");
                httpd_register_uri_handler(server, &main_uri);
                return server;
        }

        ESP_LOGI(TAG, "Error starting server!");
        return NULL;
}

根据demo写的测试代码
目前是单线程访问，刷新几次请求后就会出现以下错误
看起来像是连接数不够，理论上http短连接执行完毕后应该会自动释放连接的
但根据以上情况来看，并没有自动关闭和释放连接

[16:27:36:387] W (29257) httpd: httpd_accept_conn: error in accept (23)␍␊
[16:27:36:391] W (29267) httpd: httpd_server: error accepting new connection␍␊
[16:27:36:789] W (29667) httpd: httpd_accept_conn: error in accept (23)␍␊
[16:27:36:792] W (29667) httpd: httpd_server: error accepting new connection␍

donghengqaz, 2019-10-14:
example 里面有个 sdkconfig.defaults，你需要参考拷贝到你自己的 example 下面。

closed 2019-11-08 by FayeY

GarMingLi, 2020-07-10:
我也遇到同样的问题，请问你解决了吗？

leovs, 2020-07-24:
时间有点久了，记不清了。