espressif/ESP8266_RTOS_SDK #709: A fatal error occurred: Invalid head of packet (0x46)

jooopel, 2019-09-22:
Can someone help me with this issue? 
i tryed to find something, but i couldnt find anything.

Toolchain path: /root/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-100-ge567ec7
Compiler version: 5.2.0
Python requirements from /root/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
esptool.py v2.4.0
Connecting.....
Chip is ESP8266EX
Features: WiFi
MAC: 00:08:ff:00:00:00
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x46)
make: *** [/root/esp/ESP8266_RTOS_SDK/components/esptool_py/Makefile.projbuild:76: flash] Error 2

closed 2019-09-24 by jooopel

mavenik, 2020-03-19:
@jooopel How was this resolved?

I followed the instructions as on https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/index.html and received this error. I tried hello_world and wifi examples and I got the same error for both. I enabled trace on esptool.py and saw this output when I did a `make flash`

````
TRACE +0.000 Read invalid data: 00ff08000000000000c046
TRACE +0.000 Remaining data in serial buffer: 

A fatal error occurred: Invalid head of packet (0x46)
make: *** [flash] Error 2
````

I am using a DOIT ESP32 DevKitV1 board on a MacOS Sierra machine. Things work seamlessly if I follow the Arduino IDE approach though. So it does not seem like an issue with the board itself.

zsbukr, 2020-06-13:
Іt issue was resolved?
I have same.