espressif/ESP8266_RTOS_SDK #713: Native OTA and simple OTA

Dipendra77, 2019-09-26:
Hi,
Can anyone tell me the differences between native OTA and simple OTA support for ESP8266 and in what context of Over the Air upgrade are they used ? I just want to upgrade the firmware on the device with a new binary file?  which one should I opt for ?

Thank you in advance.

xuhongv, 2019-09-26:
simple OTA maybe is the best for you

ustccw, 2019-09-27:
@Dipendra77 please take a look at pinned issue if you are interested in `native OTA`:
https://github.com/espressif/ESP8266_RTOS_SDK/issues/489

In short, you only need to care about `simple OTA` if you set up your projects after SDK version `3.x`

Dipendra77, 2019-09-27:
> simple OTA maybe is the best for you

Can you please explain the difference ? On what basis should we decide whether to goo with simple ota or native ota?

27-301, 2019-09-29:
native ota may need to use the esp_ota, partition, http, etc. API directly, but simple ota does a wrap, so you can just call esp_https_ota, it's simple.

closed 2019-11-08 by FayeY