espressif/ESP8266_RTOS_SDK #716: Simple OTA upgrade with ESP8266  Error

Dipendra77, 2019-09-27:
Hi, 
I am trying to update my ESP8266 custom board with the new firmware using OTA upgrade. I am using the latest ESP8266_RTOS_SDK (V3.0). I am able to upgrade it with the example hello-world.bin file succesfully, but when I try to upgrade it with my own binary file I am getting this following error.
E (15034) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0xea)
E (15098) esp_https_ota: Error: esp_ota_write failed! err=0x0
E (15105) simple_ota_example: Firmware Upgrades Failed

Just to make it clear the binary file for our custom source code is compiled with the toolchain based on esp-open-sdk by pfalcon, here is the link : https://github.com/pfalcon/esp-open-sdk
and we have used esp-open-rtos ( https://github.com/SuperHouse/esp-open-rtos ) to write our firmware. 
Please find attached the Error Log screenshot.

[image]

27-301, 2019-09-29:
Sorry, don't use esp-open-sdk before, from the log, it shows that `OTA image has invalid magic byte (expected 0xE9, saw 0xea)`, maybe the binary format of esp-open-sdk is different, the first byte should be 0xE9. please check, you can try to compile examples in SDK, then do the OTA. BTW, the newest version is v3.2, not v3.0.

Dipendra77, 2019-10-02:
> Sorry, don't use esp-open-sdk before, from the log, it shows that `OTA image has invalid magic byte (expected 0xE9, saw 0xea)`, maybe the binary format of esp-open-sdk is different, the first byte [...]

Thank you for your response :)

closed 2019-10-25 by FayeY