espressif/ESP8266_RTOS_SDK #719: FOTA with ESP8266

Dipendra77, 2019-09-30:
How do we upgrade devices already in the field , all at once using the ESP8266 FOTA upgrade?

ustccw, 2019-10-09:
@Dipendra77 Please take a look at `examples/system/ota/native_ota/2+MB_flash/new_to_new_no_old` if by `HTTP`.
or `examples/system/ota/simple_ota_example` if by `HTTPS`.

ustccw, 2019-10-09:
Additional, please see as https://github.com/espressif/ESP8266_RTOS_SDK/issues/489 for details.

Dipendra77, 2019-10-10:
Hi, 
Thank you for responding. I understood the simple OTA example and can successfully upgrade to a new binary using the same. Sorry I was not very clear about the issue earlier. The actual problem is that these examples expect the device and the PC ( On which the https server is running) to be on the same network and a serial connection between the device and the PC. My use case for FOTA would be more like upgrading the firmware from a remote server ( a web server probably ) and without any physical connection between the device and the Host server. Or pushing updates through an android app and distributing it to all the devices on the field , directly to the users. 
I found this on the web which does something similar with Arduino IDE [](https://github.com/espressif/arduino-esp32/blob/master/docs/OTAWebUpdate/OTAWebUpdate.mdl)  
I wanted to know if this is possible using the RTOS_SDK for ESP8266 / ESP32 ?
Any help will be appreciated.
Thank you.

colesnicov, 2020-06-10:
For the first programming you have to connect the device to the computer with the help of a cable. You have to!  then it doesn't matter if the device connects to a remote server and downloads an upgrade from there.  The condition is that the device has access to the public Internet ..