espressif/ESP8266_RTOS_SDK #722: Bug in spiffs library

ochrin, 2019-10-07:
#100  Environment

- Development Kit:
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    release/v2.1 but should be valid for release release/v3.x
- Development Env:      Eclipse
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description
Sometime the spiffs library crashes with Fatal exception (9).

I can reproduce it because I have a spiffs image which generates some calls to some specific spiffs library function.

I don't think you will be able to reproduce it, but it does not matter as I think I know what is the problem.

After investigation, I found the following issue.
In **spiffs_gc.c**, in function **spiffs_gc_find_candidate()**
we can find this code:
```
  // using fs->work area as sorted candidate memory, (spiffs_block_ix)cand_bix/(s32_t)score
  int max_candidates = MIN(fs->block_count, (SPIFFS_CFG_LOG_PAGE_SZ(fs)-8)/(sizeof(spiffs_block_ix) + sizeof(s32_t)));
...
  s32_t *cand_scores = (s32_t *)(fs->work + max_candidates * sizeof(spiffs_block_ix));
```
And in **spiffs_config.h**:
We can find this code:
```
// Block index type. Make sure the size of this type can hold
// the highest number of all blocks - i.e. spiffs_file_system_size / log_block_size
typedef u16_t spiffs_block_ix;
```
So, unless `max_candidates` is a multiple of 2 (but I could not find any reason for), `cand_scores` may point on an address aligned on 2 bytes as `sizeof(spiffs_block_ix)`=2
This will generates a crash later on in **spiffs_gc.c**, in function **spiffs_gc_find_candidate()** in line like this:
`          cand_scores[cand_ix] = score;`

I will continue my investigation to confirm my findings and see how it can be fixed.

ochrin, 2019-10-07:
After one more check, I see this problem is solved in v3.2 with this line:
```
   // align cand_scores on s32_t boundary
  cand_scores = (s32_t*)(((intptr_t)cand_scores + sizeof(intptr_t) - 1) & ~(sizeof(intptr_t) - 1));

```

It is still here in v2.1.

donghengqaz, 2019-10-12:
Do you also use the old SDK release/v2.1?  If you need, I will add a option to fix the bug.

ochrin, 2019-10-13:
Yes, I am still using v2.1. I started my project when v3.0 was not really available and I don't want to migrate to v3.x now as I am close to the end of my project.
I will do it later.
I don't know how many people are still using v2.1, but this fix will be helpful for them.

ochrin, 2019-10-13:
By the way, I noticed an issue when using both spiffs and lwip in a source file.
If you include some lwip header files in your source, they may redefine `write()` and `read()` functions. This means that, if in your source file, you are writing or reading in a spiffs file, it may not work, as it will call lwip functions and not spiffs functions.
The solution is to remove the lwip include and move this related part of code somewhere else.

FayeY, 2019-11-08:
The solution should have been merged, please have a try and feel free to let us know if the issue is still unsolved.

closed 2019-11-08 by FayeY