espressif/ESP8266_RTOS_SDK #724: smart_config example issues

Kanson4, 2019-10-08:
使用最新SDK例程的hellow_world一切正常，但测试smart_config例程时，出现两个问题。
在make monitor中：
1、中间出现一大堆乱码
2、使用ESPTOUCH:SMARTCONFIG(version:0.3.7.1)配网时一点反应都没。
不知道什么望指教，谢谢

log如下：

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16
tail 12
chksum 0xc2
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12
tail 12
chksum 0x2b
ho 0 tail 12 room 4
load 0x3ffe8420, len 3552, room 12
tail 4
chksum 0x39
I (49) boot: ESP-IDF v3.2-180-gd6ec931e-dirty 2nd stage bootloader
I (49) boot: compile time 13:43:17
I (50) qio_mode: Enabling default flash chip QIO
I (58) boot: SPI Speed      : 40MHz
I (64) boot: SPI Mode       : QIO
I (70) boot: SPI Flash Size : 4MB
I (77) boot: Partition Table:
I (82) boot: ## Label            Usage          Type ST Offset   Length
I (93) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (105) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (117) boot:  2 factory          factory app      00 00 00010000 000f0000
I (128) boot: End of partition table
I (135) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x488a8 (297128) map
0x40210010: _flash_cache_start at ??:?

I (248) esp_image: segment 1: paddr=0x000588c0 vaddr=0x402588b8 size=0x0b47c ( 46204) map
I (264) esp_image: segment 2: paddr=0x00063d44 vaddr=0x3ffe8000 size=0x00694 (  1684) load
I (265) esp_image: segment 3: paddr=0x000643e0 vaddr=0x40100000 size=0x00a30 (  2608) load
I (275) esp_image: segment 4: paddr=0x00064e18 vaddr=0x40100a30 size=0x05674 ( 22132) load
I (295) boot: Loaded app from partition at offset 0x10000
I (342) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (350) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (415) phy_init: phy ver: 1151_0
I (419) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
SC version: V2.8.2
L C
L C╔╔D
   ╔╔D \%u@Y.+uB       -))
                          5DT\    )      5[
║╗LO@)        KTRs╔╔O5)e═@T    5
                                r
qCJiA╝Zr╔╔O5)e═@T ^)nZP-CJE  J
       Qؓ      (
               %~w╚
                   jB     JVZt7 sCQAJA
                                      %~w@SS@║1 eJ
                                                          3o[)AAPK^QR@+))Gؑ╔ xL@Y.╔    J7╔╔╔╔E
)╔╔╔1eʁ EJ     U
                %
                 ╔╚
AMi
        e╗%[
            ╔PPQV[N*
               %Y╔IT╔╔╔╔╔╔n╔╔╔╔╔XT5QKVP╔ST+)JC@╔%Y╔- ╔╔╔╔╔╔5m
╔╔╔╔,╔E╝CJAH
            %y
              ╔1o ╔╔╔╔#PNP╔╔╔╔

        %[
          ╔lT-║LXRQRY)JCB╔)YVVTL╝╔P\╔o╝╝
        %[
          ╔lT-║LXRQRY)JCB╔)YVVTL╝╔P\╔o╝╝╔j )ʃ E    m
                                                    #5lʋ1DT
                                                           XeFP59E%╔m╝XOP@
                                                                          ++)GD@╔F@JI7O 1═1\╔RRe╝
7╝╔@@
     )IJ+
        EBHAZ5T R╝E╝5\11╔m╝P# ╔EFBCH)!Eؓ  @]
                                           #5lʋ1DT
                                                  XM═5Nt╝╝5=
        EBHAZ5T R╝E╝5\11╔m╝P# ╔EFBCH)!Eؓ  @]                    )ROCJ
                                           #5lʋ1DT
                                                  XM═5Nt╝╝5=
        EBHAZ5T R╝E╝5\11╔m╝P# ╔EFBCH)!Eؓ  @]                    )ROCJC ╔%Y╔YV║P%1Ls[@
                                           #5lʋ1DT
                                                  XM═5Nt╝╝5=
                                                               )ROCJC ╔%Y╔YV║P%1Ls[@\-\ZYNI (344) s
ystem_api: Base MAC address is not set, read default base MAC address from EFUSE
I (352) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (421) phy_init: phy ver: 1151_0
I (425) reset_reason: RTC reset 4 wakeup 0 store 0, reason is 7
SC version: V2.8.2
I (3993) sc: SC_STATUS_FINDING_CHANNEL

bgkavinga, 2019-10-09:
@Kanson4 I am having this same issue. According to this https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html#watchdog documentation it is hardware restart due to application locked up.

Kanson4, 2019-10-09:
@

> @Kanson4 I am having this same issue. According to this https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html#watchdog documentation it is hardware restart due to application locked up.

thanks for you tips,i try it!

closed 2019-10-09 by Kanson4

bgkavinga, 2019-10-09:
FYI: this commit addresses this issue. https://github.com/espressif/ESP8266_RTOS_SDK/commit/d8d355bbdfa1363d43fa4f25d22de9d5b6f9f6a4

Kanson4, 2019-10-10:
> FYI: this commit addresses this issue. [d8d355b](https://github.com/espressif/ESP8266_RTOS_SDK/commit/d8d355bbdfa1363d43fa4f25d22de9d5b6f9f6a4)

it worked ,thank so much.

Simaho99, 2020-07-08:
changed the baud rate in make menu config,
//go to serial flasher config then "make monitor' baud rate (74880 bps)  --->  "

it worked!