espressif/ESP8266_RTOS_SDK #725: pwm占空比重设问题

jianggogogo, 2019-10-08:
在使用pwm来控制可控硅时，我需要保持pwm周期，相位和可控硅的周期，相位相同。因此在每个零点检测的时候，我重新设置pwm的相位和占空比。但是我发现，重新设置后采用pwm_start函数，pwm的定时器的计数器似乎并不会立即清零，而是会从刚刚的点开始继续用新的参数运行pwm。现象是：造成pwm产生对应可控硅的相移。
  后面我采用了，pwm_stop函数来实现关闭和开启，但是在高频率的重新设置pwm相位和占空比时，我又发现，pwm_stop函数是关闭这一个周期，如果在pwm_stop之后立即采用pwm_start，pwm会在这个周期停止后，才会开始新的参数。这当然会造成一个pwm周期的空白。
[image]
  这是中断回调函数里面，重新设置pwm参数的代码。在这里设置pwm相位为零，则可以保持可控硅和pwm周期相位一致。但是，这样设置之后，我发现，pwm相位对应可控硅相位，一直在移动。
这是对应的波形图：
[image]
  可以看出，相位并没有随着高电平的来到而清零。
  想请问一下，是否存在对应的函数或者说操作能够设置完pwm参数后,直接清零当前的计数（相位），然后按照新的参数执行， 或者说，此函数也是需要等待下一个周期实现更改参数。

ChenZhengwei20190741, 2019-10-09:
你的中断里有试过加`pwm_set_phases();`来改变相位吗？

ChenZhengwei20190741, 2019-10-12:
滞后时间是多长

ChenZhengwei20190741, 2019-10-12:
建议直接把pwm_start放到中断里，才不会滞后那么长。目前8266的pwm设计就是在你更改pwm占空比时，会在当前旧pwm周期结束再更改pwm占空比。

jianggogogo, 2019-10-12:
我将pwm_start放置于中断之后，可以适当的减少PWM重启相位同外部中断相位的差，但是设置PWM寄存器也需要一定时间，在高频条件下，这样的时间会被逐渐累积，最后在示波器上面的PWM同周期外部中断依然会产生相位差。这里，冒昧建议希望在下一个SDK版本开发中，提供能够立即设置PWM的函数，来更丰富开发者的操作需求，谢谢您的解答。

ChenZhengwei20190741, 2019-10-12:
是的，就是设置pwm需要一定时间，所以，你中断发生，也难以及时的改变占空比，必然会累积，特别是在高频的情况下会更明显，可能这个pwm设计的初衷应该没有想要去高频修改占空比的吧，不过我想应该后续的sdk会完善一下。

closed 2019-11-08 by FayeY