espressif/ESP8266_RTOS_SDK #727: OTA from a remote server download slowly even stop...

JunJianX, 2019-10-09:
#178  Environment

- Development Kit: xtensa-lx106
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):  ESP8266_RTOS_SDK
- Development Env:      Make
- Operating System:     Ubuntu
- Power Supply:         USB

## Problem Description

//Detailed problem description goes here.
I run "**ota_example_main**" ,when I from my local computer(my pc as server ) ,I finished OTA quickly.But when I download from a url(on IOT platform),speed get slowly and even stop last.
It bothered me a long time ,Could anyone have ideas?THANKS.

## Debug Logs

connected with Xiaomi_test, channel 9
[0;32mI (10061) event: sta ip: 192.168.31.67, mask: 255.255.255.0, gw: 192.168.31.1[0m
[0;32mI (10061) ota: Connect to Wifi ! Start to Connect to Server....[0m
[0;32mI (10061) ota: Server IP: 39.156.46.55 Server Port:80[0m
AF_INET and SOCK_STREAM IS:2 1

[0;32mI (10101) ota: Connected to server[0m
[0;32mI (10101) ota: Connected to http server[0m
[0;32mI (10101) ota: Send GET request to server succeeded[0m
[0;32mI (10111) ota: Writing to partition subtype 17 at offset 0x110000[0m
[0;32mI (16231) ota: esp_ota_begin succeeded[0m
[0;32mI (16231) ota: Totoal OTA number 2 update to 1 part[0m
[0;32mI (16241) ota: parse Content-Length:382608, ota_size 382608[0m
[0;32mI (16241) ota: Have written image length 1024[0m
[0;32mI (16251) ota: Have written image length 2048[0m
[0;32mI (16261) ota: Have written image length 3072[0m
[0;32mI (16271) ota: Have written image length 4096[0m
[0;32mI (16281) ota: Have written image length 5069[0m
[0;32mI (16301) ota: Have written image length 6093[0m
[0;32mI (16311) ota: Have written image length 7117[0m
[0;32mI (16321) ota: Have written image length 8141[0m
[0;32mI (16331) ota: Have written image length 9165[0m
[0;32mI (16331) ota: Have written image length 9425[0m
[0;32mI (16341) ota: Have written image length 10449[0m
[0;32mI (16351) ota: Have written image length 10877[0m
[0;32mI (16391) ota: Have written image length 11901[0m
[0;32mI (16401) ota: Have written image length 12329[0m
[0;32mI (16701) ota: Have written image length 12361[0m
[0;32mI (16921) ota: Have written image length 13385[0m
[0;32mI (16931) ota: Have written image length 14409[0m
[0;32mI (16941) ota: Have written image length 15433[0m
[0;32mI (16941) ota: Have written image length 16457[0m
[0;32mI (16951) ota: Have written image length 16685[0m
[0;32mI (17521) ota: Have written image length 17709[0m
[0;32mI (17531) ota: Have written image length 18137[0m
[0;32mI (17641) ota: Have written image length 18169[0m
[0;32mI (17651) ota: Have written image length 19193[0m
[0;32mI (17651) ota: Have written image length 19589[0m
[0;32mI (17671) ota: Have written image length 20613[0m
[0;32mI (17681) ota: Have written image length 21637[0m
[0;32mI (17681) ota: Have written image length 22493[0m
[0;32mI (18851) ota: Have written image length 23517[0m
[0;32mI (18851) ota: Have written image length 23945[0m
[0;32mI (18971) ota: Have written image length 24969[0m
[0;32mI (18971) ota: Have written image length 25397[0m
[0;32mI (18991) ota: Have written image length 26421[0m
[0;32mI (19001) ota: Have written image length 27445[0m
[0;32mI (19011) ota: Have written image length 28301[0m
[0;32mI (21311) ota: Have written image length 29325[0m
[0;32mI (21311) ota: Have written image length 29753[0m
[0;32mI (21401) ota: Have written image length 29785[0m
[0;32mI (21411) ota: Have written image length 30809[0m
[0;32mI (21421) ota: Have written image length 31205[0m
[0;32mI (21431) ota: Have written image length 32229[0m
[0;32mI (21441) ota: Have written image length 33253[0m
[0;32mI (21441) ota: Have written image length 34109[0m
[0;32mI (26041) ota: Have written image length 35133[0m
[0;32mI (26051) ota: Have written image length 35561[0m
[0;32mI (26171) ota: Have written image length 35593[0m
[0;32mI (26191) ota: Have written image length 36617[0m
[0;32mI (26191) ota: Have written image length 37013[0m
[0;32mI (26211) ota: Have written image length 38037[0m
[0;32mI (26211) ota: Have written image length 39061[0m
[0;32mI (26231) ota: Have written image length 39917[0m
[0;32mI (35401) ota: Have written image length 40941[0m
[0;32mI (35411) ota: Have written image length 41369[0m
[0;32mI (35651) ota: Have written image length 41401[0m
[0;32mI (35661) ota: Have written image length 42425[0m
[0;32mI (35671) ota: Have written image length 43449[0m
[0;32mI (35681) ota: Have written image length 44473[0m
[0;32mI (35691) ota: Have written image length 45497[0m
[0;32mI (35691) ota: Have written image length 45725[0m
[0;32mI (54151) ota: Have written image length 46749[0m
[0;32mI (54151) ota: Have written image length 47177[0m
[0;32mI (54411) ota: Have written image length 47209[0m
[0;32mI (54661) ota: Have written image length 48233[0m
[0;32mI (54661) ota: Have written image length 48629[0m
[0;32mI (54901) ota: Have written image length 49653[0m
[0;32mI (54911) ota: Have written image length 50677[0m
[0;32mI (54921) ota: Have written image length 51533[0m

ustccw, 2019-10-09:
@JunJianX It looks like a http server issue. Would you mind providing the `OTA URL` for us to test.

JunJianX, 2019-10-10:
@ustccw Thanks..    https://cnbj2.fds.api.xiaomi.com/ks-firmware-dev/aws8266/01.bin

ustccw, 2019-10-11:
@JunJianX it looks like work well on my side.
[image]

**Maybe you could try the following tips:**
1. Free heap
please make sure free heap is more that `10KB+`

2. Remove OTA irrelevant code
please test with `examples/system/ota/native_ota/2+MB_flash/new_to_new_no_old` for simply issue.

3. Try the latest release version
`release/v3.2` is recommended, commitID is `cdcfbee8a488a5e10ef1504d126920ac43fe898d`

4. Set 802.11b/g mode for ESP8266 for avoiding router compatibility problems
```
// ... WiFi initialise
esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCAL_11B | WIFI_PROTOCAL_11G);
```

5. Capture air packets and see what happened
normally, it is easily for us to see what happened

Dipendra77, 2019-10-14:
Hi JunJianX,
I am trying to do OTA update from a remote https server running on my host pc with native_ota example located at  examples/system/ota/native_ota/2+MB_flash/new_to_new_no_old. I am able to perform update when the HTTP server and the device are on the same network. But when I try with the device connected on a network different than what the HTTP server is running on , I am getting following errors as :
E (21852) ota: Error: receive data error! errno=104
E (21856) ota: Exiting task due to fatal error...

[image]

Looks like you were able to achieve OTA from a remote server. Can you be please share the solution ?

Thanks in advance.

JunJianX, 2019-10-15:
@Dipendra77  In the README.md file.

## Error "ota_begin error err=0x104"

If you see this error then check that the configured (and actual) flash size is large enough for the partitions in the partition table. The default "two OTA slots" partition table only works with 4MB flash size. To use OTA with smaller flash sizes, create a custom partition table CSV (look in components/partition_table) and configure it in menuconfig.

If changing partition layout, it is usually wise to run "make erase_flash" between steps.

You can find your answer!

closed 2019-11-08 by FayeY