espressif/ESP8266_RTOS_SDK #730: wifi APSTA mode - disconnects and reboots

Dipendra77, 2019-10-11:
Hi I am trying to run a simple in APSTA mode But it does not seem to be stable. it disconnects from the AP and the chip resets every few seconds. Below is the sample code. Can anyone tel me how to establish a stable connection in APSTA Mode?

void wifi_init()
{

wifi_event_group = xEventGroupCreate();
tcpip_adapter_init();
ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL) );

wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_wifi_init(&cfg));

strncpy((char*)sta_config.sta.ssid, WIFI_SSID, 32);
strncpy((char*)sta_config.sta.password, WIFI_PASS, 64);

wifi_config_t ap_config = {
.ap = {
.ssid = AP_SSID,
.ssid_len = strlen(AP_SSID),
.max_connection = 3,
.authmode = WIFI_AUTH_OPEN,
.beacon_interval = 0;

},
};

ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &sta_config));
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &ap_config));
ESP_ERROR_CHECK(esp_wifi_start());
ESP_LOGI(WIFI_TAG, "wifi_init finished.SSID:%s password:%s",WIFI_SSID, WIFI_PASS);

}

donghengqaz, 2019-10-12:
Could you supply your full examples to me? Because I can't find the local variable `sta_config`, so I can't compile and test it.

Dipendra77, 2019-10-14:
Thank you for your response. It got fixed. I found out there is some issue in the serial port of my laptop which causes random restart of the device. Works fine on the other laptop.

closed 2019-10-14 by Dipendra77