espressif/ESP8266_RTOS_SDK #736: 更新了master分支，编译工程，mbedtls 报错 ，无法编译！

xuhongv, 2019-10-21:
## Environment

- Development Kit:
- IDF version:  39c8f458
- Development Env:     Make
- Operating System:     Windows
- Power Supply:        external 5V

## Problem Description

编译报错：

```
CC build/mbedtls/mbedtls/library/ssl_ticket.o
E:/Espressif/Esp32/esp-idf/home/XuHongYss/ESP8266_RTOS_SDK/components/mbedtls/mbedtls/library/ssl_ticket.c: In function 'mbedtls_ssl_ticket_write':
E:/Espressif/Esp32/esp-idf/home/XuHongYss/ESP8266_RTOS_SDK/components/mbedtls/mbedtls/library/ssl_ticket.c:345:17: error: implicit declaration of function 'mbedtls_cipher_auth_encrypt' [-Werror=implicit-function-declaration]
     if( ( ret = mbedtls_cipher_auth_encrypt( &key->ctx,
                 ^
E:/Espressif/Esp32/esp-idf/home/XuHongYss/ESP8266_RTOS_SDK/components/mbedtls/mbedtls/library/ssl_ticket.c: In function 'mbedtls_ssl_ticket_parse':
E:/Espressif/Esp32/esp-idf/home/XuHongYss/ESP8266_RTOS_SDK/components/mbedtls/mbedtls/library/ssl_ticket.c:436:17: error: implicit declaration of function 'mbedtls_cipher_auth_decrypt' [-Werror=implicit-function-declaration]
     if( ( ret = mbedtls_cipher_auth_decrypt( &key->ctx, iv, 12,
                 ^
E:/Espressif/Esp32/esp-idf/home/XuHongYss/ESP8266_RTOS_SDK/components/mbedtls/mbedtls/library/ssl_ticket.c: At top level:
cc1.exe: warning: unrecognized command line option '-Wno-frame-address'
cc1.exe: some warnings being treated as errors
make[1]: *** [/home/XuHongYss/ESP8266_RTOS_SDK/make/component_wrapper.mk:292：mbedtls/library/ssl_ticket.o] 错误 1
make: *** [E:/Espressif/Esp32/esp-idf/home/XuHongYss/ESP8266_RTOS_SDK/make/project.mk:571：component-mbedtls-build] 错误 2
```
- 下面是我的 mbedtls 的配置
[image]

Kanson4, 2019-10-21:
I have same issues.

Alexlcb, 2019-11-14:
怎么没人回复呢

xuhongv, 2019-11-14:
@Kanson4 @Alexlcb 这个问题我已经解决了，通过下载最新的  gui 环境就可以了，我猜到你们应该是 2018年的环境把；

Alexlcb, 2019-11-19:
> 
> 
> @Kanson4 @Alexlcb 这个问题我已经解决了，通过下载最新的 gui 环境就可以了，我猜到你们应该是 2018年的环境把；

我这个是在linux环境编译的，刚开始是simplewifi这个例程有bug，我git pull后就出现这个问题了。请问linux环境怎么修复？

xuhongv, 2019-11-19:
linux环境我没弄过，你重新克隆全部看看？

xuhongv, 2019-11-21:
- 找到原因了，旧版的 工程的 sdkconfig 文件的某个变量配置导致这个库无法正常编译；
- 解决方法：重新把当前sdk随便一个 sdkconfig 、Makefile文件复制替换当前工程的，以及删除 build 文件夹所有编译子文件；重新编译即可！

closed 2019-11-21 by xuhongv