espressif/ESP8266_RTOS_SDK #737: OTA old sdk -> new sdk doesn't work for me

berkutta, 2019-10-21:
I tried to perform an OTA from old sdk to new sdk but I can't get it working. I even tried it now via esptool to be sure that the OTA routine isn't the issue:

```
# ./flash.sh && picocom -b 115200 /dev/ttyUSB2
esptool.py v2.3.2-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Compressed 4080 bytes to 2936...
Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.3 seconds (effective 124.4 kbit/s)...
Hash of data verified.
Compressed 300468 bytes to 214565...
Wrote 300468 bytes (214565 compressed) at 0x00001000 in 18.8 seconds (effective 127.8 kbit/s)...
Hash of data verified.
Compressed 300468 bytes to 214606...
Wrote 300468 bytes (214606 compressed) at 0x00081000 in 18.8 seconds (effective 127.8 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 83.4 kbit/s)...
Hash of data verified.
Compressed 164 bytes to 99...
Wrote 164 bytes (99 compressed) at 0x000fd000 in 0.0 seconds (effective 91.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

picocom v3.1

port is        : /dev/ttyUSB2
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready
���c
    l
     �b�l{d���g�s��o|�
                      l
                       ddd`�rn��`SDK version:2.0.0(e271380) 
App version:esp8266-sensor-201806131321 
--------------------------------------
SSID:

[...]

# ~/esp/esptool.py --port /dev/ttyUSB2 write_flash 0x00001000 ota.v2_to_v3.ota.bin && picocom -b 74880 /dev/ttyUSB2 
esptool.py v2.3.2-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 336996 bytes to 223156...
Wrote 336996 bytes (223156 compressed) at 0x00001000 in 19.6 seconds (effective 137.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

picocom v3.1

port is        : /dev/ttyUSB2
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

I (85) boot: ESP-IDF v3.2-214-g92f9b7c9e 2nd stage bootloader
I (85) boot: compile time 15:05:17
I (94) qio_mode: Enabling default flash chip QIO
I (94) boot: SPI Speed      : 40MHz
I (99) boot: SPI Mode       : QIO
I (105) boot: SPI Flash Size : 4MB
E (112) partition_port: read V2 system param error 1
E (120) partition_port: flash read parameters is not align, value is 0x3ffe992c 1000 ffffffff
E (135) partition_port: read flash 4096 bytes from 0xffffffff error
E (145) boot: Failed to update partition table location
E (154) boot: load partition table error!

```

FayeY, 2019-11-08:
May I ask if there is update? Is the issue solved now?

berkutta, 2019-11-08:
@FayeY Thanks for asking, the issue is still persisting. I flashed the OTA binary over esptool to rule out OTA issues. But it still doesn't work.

berkutta, 2019-11-13:
The same issue happens no matter if I flash the new SDK OTA bin to 0x01000 or 0x81000. What do I need to change to get OTA from old to new SDK working?

berkutta, 2019-11-26:
Any news? Will we ever get working OTA for old -> new SDK for products which we already deployed to customers?

berkutta, 2020-07-30:
@FayeY Any news? Also pack_fw.py not working with python3.

```
/home/user/esp/ESP8266_RTOS_SDK/tools/pack_fw.py:110: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if app_offset is not 0:
/home/user/esp/ESP8266_RTOS_SDK/tools/pack_fw.py:118: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if app_offset is 0:
pack_fw.py v1.0.2
Traceback (most recent call last):
  File "/home/user/esp/ESP8266_RTOS_SDK/tools/pack_fw.py", line 196, in <module>
    _main()
  File "/home/user/esp/ESP8266_RTOS_SDK/tools/pack_fw.py", line 190, in _main
    main()
  File "/home/user/esp/ESP8266_RTOS_SDK/tools/pack_fw.py", line 169, in main
    operation_func(args)
  File "/home/user/esp/ESP8266_RTOS_SDK/tools/pack_fw.py", line 104, in pack3
    fw_data += data
TypeError: can only concatenate str (not "bytes") to str
make: *** [/home/user/esp/ESP8266_RTOS_SDK/components/esp8266/Makefile.projbuild:148: build/sw_esp8266_environmental_sensing.v2_to_v3.ota.bin] Error 1
```