espressif/ESP8266_RTOS_SDK #739: ESP8266_RTOS_SDK 2.0，重新编译LWIP库出问题

zouzhe1, 2019-10-27:
1，由于HiLink SDK 适配网络接口需要设置socket 的SO_REUSEADDR 属性，而乐鑫官方
SDK 2.0 的LWIP 配置中未使能SO_REUSEADDR 属性，因此需要修改LWIP 的配置。
修改方法如下：
在ESP8266_RTOS_SDK-2.0.0/include/lwip 目录下，修改lwipopts.h 文件中的宏定义
SO_REUSE 的值，由0 改为1。
2，我修改后，在ESP8266_RTOS_SDK-2.0.0/third_party 目录下，执行make_lib.sh 脚本，命令如
下：sh make_lib.sh lwip，然后出错
[image]

3，看这个出错的意思是好像没有找到sockets.o

4, 我找到sockets.c文件，看到开头有个宏判断，
`#if LWIP_SOCKET /* don't build if not configured for use in lwipopts.h */`
然后我在lwipopts.h文件里没有找到这个宏，我手动加上这个宏得1，编译还是这样，没有变化。

5，请问这是什么原因呢？谢谢

FayeY, 2019-11-08:
想请问，这个问题有后续进展么，是否已经解决？

FayeY, 2020-01-06:
由于长时间无更新，此问题暂时关闭。如有需要，欢迎 reopen issue，继续讨论。谢谢！

closed 2020-01-06 by FayeY

woodyfan, 2020-02-02:
类似问题，我在做相同操作时遇到了这个问题。
In file included from igmp.c:87:0:
**igmp.c: In function 'igmp_stop':
/mnt/hgfs/hilink/Hilink-HN/ESP8266_RTOS_SDK-2.1.1/include/lwip/lwip/mem.h:81:3: error: implicit declaration of function 'vPortFree' [-Werror=implicit-function-declaration]**
   vPortFree(s, file, __LINE__);\
   ^
找到vPortfree函数位置，并在mem.h中添加如下include并不能解决问题
`#include "../../freertos/mpu_wrappers.h"`