espressif/ESP8266_RTOS_SDK #741: Bugs / problem in ESP8266_RTOS_SDK  v3.2    /examples/wifi/simple_wifi  program in AP mode

sajan-p-jose, 2019-10-28:
## Environment

- Development Kit:
- IDF version (RTOS_SDK  v3.2): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:     Make
- Operating System:    Ubuntu
- Power Supply:        external 3.3V

## Problem Description
AP mode Wifi not connecting, program not stable , restarting
Compiled ESP8266_RTOS_SDK  v3.2    /examples/wifi/simple_wifi  sample program from SDK but it is not working properly
problem can't detect from console log

### Expected Behavior
Wifi connection in AP mode
### Actual Behavior
Wifi not connecting , system restaring
### Steps to repropduce

1. step1
Compile and run ESP8266 IDF V3.2  /examples/wifi/simple-example program
without any modifications 

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
/* Simple WiFi Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "rom/ets_sys.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"

/* The examples use simple WiFi configuration that you can set via
   'make menuconfig'.

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_MODE_AP   CONFIG_ESP_WIFI_MODE_AP //TRUE:AP FALSE:STA
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_MAX_STA_CONN       CONFIG_MAX_STA_CONN

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
const int WIFI_CONNECTED_BIT = BIT0;

static const char *TAG = "simple wifi";

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
[... 168 lines cut ...]
D (756) nvs: nvs_get ap.max.conn 1
D (761) nvs: nvs_get bcn.interval 2
D (766) nvs: nvs_get ap.phym 1
D (771) nvs: nvs_get ap.phybw 1
D (775) nvs: nvs_get ap.sndchan 1
D (780) nvs: nvs_get lorate 1
V (785) wifi: index=19 value=0x3ffe90a0 flag=1
D (789) nvs: nvs_set_blob ap.ssid 36
V (794) wifi: index=21 value=mypassword flag=1
D (798) nvs: nvs_set_blob ap.passwd 65
V (804) wifi: index=24 value=4 flag=1
D (808) nvs: nvs_set ap.authmode 1 4
V (812) wifi: index=23 value=1 flag=1
D (816) nvs: nvs_set ap.chan 1 1
V (820) wifi: index=25 value=0 flag=1
D (823) nvs: nvs_set ap.hidden 1 0
V (828) wifi: index=26 value=4 flag=1
D (831) nvs: nvs_set ap.max.conn 1 4
V (836) wifi: index=27 value=100 flag=1
«A,P1,%e¢»í)(Hl!%y.!y¨v(p-û~1ôQèqht-·ít1R11yÐ9"!Ûo@«r«òÊI-9¨iÎΦôy(rz«y«r5         ­E
## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

donghengqaz, 2019-10-30:
Did you use the raw example ? If so, you can supply your sdkconfig to me, and then I will check this.

sajan-p-jose, 2019-10-30:
Yes, Problem on raw example in 
AP mode
 sdkconfig attached

[sdkconfig.txt](https://github.com/espressif/ESP8266_RTOS_SDK/files/3789089/sdkconfig.txt)

sajan-p-jose, 2020-04-01:
Not applicable for new version

closed 2020-04-01 by sajan-p-jose