espressif/ESP8266_RTOS_SDK #742: 从2.0版本SDK过来使用当前最新的SDK，怎么样生成app1 app2 设置flash map?

Yvan623048750, 2019-10-29:
升级方式不再是APP1 APP2 交叉升级的方式了？
设置flash map?
新的SDK为何一点文档说明都没有？
目前编译只生成了一个bin

用户区，代码区分区方式是不是整个变了？

FayeY, 2019-10-29:
烦请看看这几个文档，是否有帮助
https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/?badge=latest
https://github.com/espressif/ESP8266_RTOS_SDK/issues/489

Yvan623048750, 2019-10-29:
提问的问题都解决了，现在还有一个疑问，下载工具里各个文件的下载地址是什么？
boot_v1.6.bin                                  0x0
esp_init_data_default.bin                0x1FC000
blank.bin                                         0x1FE000
user1.2048.new.3.bin                      0x01000
user2.2048.new.3.bin                      0x81000

FayeY, 2019-11-08:
抱歉，没有太理解这个问题，是指固件如何生成吗？固件一般通过 make 指令编译生成，make flash 指令可以直接烧录。详细可以参考 https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/index.html

FayeY, 2020-01-06:
由于无更新，此问题暂时关闭。如有需要，欢迎 reopen issue，继续讨论。谢谢！

closed 2020-01-06 by FayeY