espressif/ESP8266_RTOS_SDK #743: When SPIFFS Enabled => coap_io_socket.c:162:5: error: implicit declaration of function 'close'

ricardoper, 2019-10-29:
Hello,

- ESP8266 (1MB Flash) And ESP8285 (1MB Flash)
- Release v3.2 (3ad3fd78)
- Spiffs Example (https://github.com/espressif/ESP8266_RTOS_SDK/tree/release/v3.2/examples/storage/spiffs)

When I enable SPIFFS at "menuconfig", like indicate instructions, the build fails in CoAP component:

```
Toolchain path: /esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: crosstool-ng-1.22.0-92-g8facf4c
Compiler version: 5.2.0
GENCONFIG
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/spi_flash/src/spi_flash.o
CC build/bootloader/spi_flash/port/port.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/esp8266/source/ets_printf.o
AR build/bootloader/esp8266/libesp8266.a
CC build/bootloader/util/src/sha.o
CC build/bootloader/util/src/base64.o
CC build/bootloader/util/src/aes.o
AR build/bootloader/util/libutil.a
LD build/bootloader/bootloader.elf
esptool.py v2.4.0
Building partitions from /esp8266/ESP8266_RTOS_SDK/components/partition_table/partitions_singleapp.csv...
CC build/app_update/esp_ota_ops.o
AR build/app_update/libapp_update.a
CC build/bootloader_support/src/bootloader_flash.o
CC build/bootloader_support/src/flash_qio_mode.o
CC build/bootloader_support/src/efuse.o
CC build/bootloader_support/src/bootloader_init.o
CC build/bootloader_support/src/bootloader_common.o
CC build/bootloader_support/src/secure_boot.o
CC build/bootloader_support/src/bootloader_clock.o
CC build/bootloader_support/src/bootloader_random.o
CC build/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader_support/src/flash_partitions.o
CC build/bootloader_support/src/esp_image_format.o
CC build/bootloader_support/src/bootloader_sha.o
CC build/bootloader_support/src/flash_encrypt.o
CC build/bootloader_support/src/bootloader_utility.o
AR build/bootloader_support/libbootloader_support.a
CC build/coap/libcoap/src/address.o
CC build/coap/libcoap/src/async.o
CC build/coap/libcoap/src/block.o
CC build/coap/libcoap/src/coap_time.o
CC build/coap/libcoap/src/debug.o
CC build/coap/libcoap/src/encode.o
CC build/coap/libcoap/src/mem.o
CC build/coap/libcoap/src/net.o
CC build/coap/libcoap/src/option.o
CC build/coap/libcoap/src/pdu.o
CC build/coap/libcoap/src/resource.o
CC build/coap/libcoap/src/str.o
CC build/coap/libcoap/src/subscribe.o
CC build/coap/libcoap/src/uri.o
CC build/coap/port/coap_io_socket.o
/esp8266/ESP8266_RTOS_SDK/components/coap/port/coap_io_socket.c: In function 'coap_new_endpoint':
/esp8266/ESP8266_RTOS_SDK/components/coap/port/coap_io_socket.c:162:5: error: implicit declaration of function 'close' [-Werror=implicit-function-declaration]
     close (sockfd);
     ^
cc1: some warnings being treated as errors
/esp8266/ESP8266_RTOS_SDK/make/component_wrapper.mk:285: recipe for target 'port/coap_io_socket.o' failed
make[1]: *** [port/coap_io_socket.o] Error 1
/esp8266/ESP8266_RTOS_SDK/make/project.mk:517: recipe for target 'component-coap-build' failed
make: *** [component-coap-build] Error 2
```

Regards,
Ricardo Pereira.

donghengqaz, 2019-10-30:
You may pull and use the branch release/v3.2，we have pushed some bugfix there.

ricardoper, 2019-10-30:
Hello @donghengqaz ,

If a follow the docs:
_"git clone -b v3.2 --recursive https://github.com/espressif/ESP8266_RTOS_SDK.git"_

_"The git clone option -b v3.2 tells git to clone the tag in the ESP8266_RTOS_SDK repository corresponding to this version of the documentation."_

I thought you were moving the version of this tag, if not what is the point of indicating a version that never moves.

**Suggestion**:
Is better to clone/checkout the v3.2 branch instead.

About this "issue", I'm checkout branch `v3.2` (efd0a545) and problem fixed.

Thanks for your fastest response.

Regards,
Ricardo Pereira.

closed 2019-10-30 by ricardoper