espressif/ESP8266_RTOS_SDK #744: In ESP8285 gpio_set_direction not working

ricardoper, 2019-10-29:
Hello,

- ESP8285 (1MB Flash - SPI Mode: DOUT)
- Release v3.2 (3ad3fd7)

In ESP8285, the function `gpio_set_direction` doesn't works like indicate the docs ( https://docs.espressif.com/projects/esp8266-rtos-sdk/en/v3.2/api-reference/peripherals/gpio.html#_CPPv418gpio_set_direction10gpio_num_t11gpio_mode_t).

Example:
```cpp
#include <driver/gpio.h>

void app_main(void)
{
    gpio_set_direction(LED_PIN, GPIO_MODE_OUTPUT);
    gpio_set_level(LED_PIN, 1);
}
```

LED never lights but if I write this code:

```cpp
#include <driver/gpio.h>

void app_main(void)
{
    gpio_config_t io_out_conf = {
            .intr_type = GPIO_INTR_DISABLE,
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = (1ULL<<LED_PIN),
            .pull_down_en = 0,
            .pull_up_en = 0,
    };
    gpio_config(&io_out_conf);

    gpio_set_level(LED_PIN, 1);
}
```

LED lights as expected.

Regards,
Ricardo Pereira.

donghengqaz, 2019-10-30:
You should initialize the GPIO hardware like the second example, just configure one attribute may not work well.

ricardoper, 2019-10-30:
Hello @donghengqaz ,

Sorry but I disagree in this point. This only fails in Chip ESP8285.

I'm working with Chips ESP32, ESP8266, ESP8285 more than two years and this problem only occurs with ESP8285.

Checkout branch v3.2 (efd0a54) and thhe problem remains.

Regards,
Ricardo Pereira.

xiongyumail, 2019-11-08:
Perhaps 8285 and 8266 have different IO initialization functions in ROM code.

superddr, 2022-12-29:
I have the same issue with esp32:  gpio_set_direction(12, GPIO_MODE_OUTPUT) not working, but works on pin27 26 25. not working on 12 13 14. 
gpio_config() works fine....