espressif/ESP8266_RTOS_SDK #745: 1MB Flash OTA's Errors

ricardoper, 2019-10-29:
Hello,

- ESP8266 (1MB Flash) And ESP8285 (1MB Flash)
- Release v3.2 (3ad3fd7)

Maybe this is more "I need help" than an "Issue" but I still fighting for days to see a OTA working in a 1M Flash device. I got a lot of errors, progressing very slowly but no results.

The OTA RTOS SDK examples are bad, the descriptions are no sense in many cases and code for me can be better.

The example that give me better results is "[new_to_new_no_old_copy](https://github.com/espressif/ESP8266_RTOS_SDK/tree/release/v3.2/examples/system/ota/native_ota/1MB_flash/new_to_new_no_old_copy)", but I still freeze in this error "`**Important**: The OTA binary link data is error`".

After investigating I went to the origin of the error and I still thinking in this block code locates in `app_update/esp_ota_ops.c` line 72:
```cpp
#ifndef CONFIG_ESP8266_BOOT_COPY_APP
static inline int esp_ota_verify_binary(const esp_partition_pos_t *pos, esp_image_header_t *image)
{
    const int32_t entry = image->entry_addr - 0x40200010;

    ESP_LOGD(TAG, "OTA binary start entry 0x%x, partition start from 0x%x to 0x%x\n", entry, pos->offset,
                        pos->offset + pos->size);

    if (pos->offset + pos->size <= 0x100000) {
        if (entry <= 0 || entry <= pos->offset || entry >= pos->offset + pos->size) {
            const char *doc_str = "<<ESP8266_RTOS_SDK/examples/system/ota/README.md>>";

            ESP_LOGE(TAG, "**Important**: The OTA binary link data is error, "
                          "please refer to document %s for how to generate OTA binaries", doc_str);
            return ESP_ERR_INVALID_ARG;
        }
    }

    return ESP_OK;
}
#endif
```

I'm a newbie and for me it's hard to understand some points, but this block reject everything that have a size smaller or equal than 1M, ¿right?

If is this right, how can I do a OTA for device with 1M Flash or less like ESP8285 or ESP8266?

Do you have an OTA library that works well? Can someone shares a good project?

Regards,
Ricardo Pereira.

donghengqaz, 2019-10-30:
First you may read the documents of the OTA example and its technology instroduction, the target OTA firmware of 1MB or 2MB flash is dirrerent, so you shoud configure the flash size at "make menuconfig".

ricardoper, 2019-10-30:
Hello @donghengqaz ,

I config Flash size of 1M and the correct partition tables. I tried a lot of situations and I used custom partitions too with and without spiffs partition and the result is always the same. Error "`**Important**: The OTA binary link data is error`".

The result of this sum `(pos->offset + pos->size)` for my device (`ESP8285`) is always the same, **1048576 (DEC)** => **0x100000**

The result is the same size of my flash size, so if you see the code 1MB Flash it will not be accepted:
```cpp
if (pos->offset + pos->size <= 0x100000) {
```

This week will be imposible to me but in the next week I will read all de docs again to see what I'm doing wrong but at this moment I see something wrong in this code block.

Thanks for you support and fastest responses,
Ricardo Pereira.

copercini, 2019-11-22:
I'm getting the same issue with a custom partition table and 1MB of flash: `E (32101) esp_ota_ops: **Important**: The OTA binary link data is error, please refer to document <<ESP8266_RTOS_SDK/examples/system/ota/README.md>> for how to generate OTA binaries`

Partitions:
```
# Name,   Type, SubType, Offset,   Size
nvs,      data, nvs,     0x9000,   0x4000
otadata,  data, ota,     0xd000,   0x2000
phy_init, data, phy,     0xf000,   0x1000
ota_0,    0,    ota_0,   0x10000,  0x78000
ota_1,    0,    ota_1,   0x88000, 0x78000
```
Please let me know if you discover something

copercini, 2019-11-22:
I fixed by using `CONFIG_ESP8266_BOOT_COPY_APP=y` on make menuconfig

ricardoper, 2019-11-23:
With this fix you are bypassing the `esp_ota_verify_binary` and this will fix the problem, but is this the best way to do this?

Isn't there a better way to support 1MB Flash avoid bypassing the OTA binary verification?

Changing the code (see my first post) from:
`if (pos->offset + pos->size <= 0x100000) {`

To:
`if (pos->offset + pos->size < 0x100000) {`

Is it not a more appropriate solution to this problem?

Regards.

ricardoper, 2019-11-23:
Some partitions tables:

```
# Name,    Type,  SubType,  Offset,   Size
nvs,       data,  nvs,      0x9000,   16K
otadata,   data,  ota,      0xd000,   8K
phy_init,  data,  phy,      0xf000,   4K
ota_0,     app,   ota_0,    0x10000,  448K
ota_1,     app,   ota_1,           ,  448K
spiffs,    data,  spiffs,          ,   64K
```

```
#   Name,  Type,  SubType,   Offset,     Size
     nvs,  data,      nvs,   0x9000,   0x4000
 otadata,  data,      ota,   0xd000,   0x2000
phy_init,  data,      phy,   0xf000,   0x1000
   ota_0,     0,    ota_0,  0x10000,  0x60000
   ota_1,     0,    ota_1,  0x80000,  0x60000
 storage,  data,   spiffs,  0xe0000,  0x20000
```

copercini, 2019-11-23:
yeah, `CONFIG_ESP8266_BOOT_COPY_APP=y` is not the best fix for it, as this will copy the downloaded .bin to both partitions, and if this is broken, both OTA partitions will be with problems

I'll try your solution

uddhavdave1, 2020-01-27:
Hi, facing the same issue with 1MB flash size.
I believe in my case the condition` entry >= pos->offset + pos->size` is coming to be true and causing the interruption.
when I bypassed this whole condition it gave me fatal exception(0). It would be great if anyone explained how this is happening. I am a newbie to esp programming.

uddhavdave1, 2020-02-03:
Ok, I figured my error was due to image entry not being between the ota_1 partition. This issue was solved by making app2.bin for the firmware using `make ota`  
So, print out the image entry address and use the respective app binary from make ota for concerned ota partition. 
Moreover, this is an easy workaround and a permanent solution needs to be addressed by espressif devs to make the necessary changes.

mriksman, 2020-03-27:
Bump.
I'm seeing this issue. I'm certainly no expert, but shouldn't the OTA of an ESP8266 be similar to ESP32 now? Using ESP8266 RTOS SDK, shouldn't the process be;

- Create a partition with `ota_0` and `ota_1`
- Load (via serial) an application into `ota_0`
- Create the `hello_world` example, and simply `idf build` to create the `bin` file
- Use the application in `ota_0` to load the `hello_world.bin` into `ota_1`

What is this `entry` value? What is this `esp_ota_verify_binary` trying to achieve?

mriksman, 2020-03-27:
> I fixed by using `CONFIG_ESP8266_BOOT_COPY_APP=y` on make menuconfig

This 'fix' copies the `ota_1` to `ota_0` - so the entry point is correct when it boots from `ota_0`.