espressif/ESP8266_RTOS_SDK #746: ESP MQTT reconnect error mbedtls ssl

sky9971, 2019-10-30:
I'm using ESP mqtt library for establishing ssl connection with my server. For ssl handshake i'm using mebedtls.

Along with esp mqtt i'm using esp http server.

For the first time the device connects successfully but the mqtt connection disconnects by throwing a **NO_PING_RESPONSE error**.
After that ESP_MQTT library tries to re connect but it fails to establish a connection.
It throws a **mbedtls handshake error -0x4310** 
After this I have to restart the device to for it to connect to my cloud.

Kindly help me resolve this issue.

donghengqaz, 2019-11-04:
Could you show me your log? You may open the mbedTLS debugging information at menuconfig. It will help to sovle the problem more effectively.

sky9971, 2019-11-04:
`I (3900) Sky9971-main: Free memory: 54608 bytes
mbedtls: ssl_tls.c:6674 => handshake
mbedtls: ssl_cli.c:3336 client state: 0
mbedtls: ssl_tls.c:2464 => flush output
mbedtls: ssl_tls.c:2476 <= flush output
mbedtls: ssl_cli.c:3336 client state: 1
mbedtls: ssl_tls.c:2464 => flush output
mbedtls: ssl_tls.c:2476 <= flush output
mbedtls: ssl_cli.c:730 => write client hello
mbedtls: ssl_cli.c:768 client hello, max version: [3:3]
mbedtls: ssl_cli.c:706 client hello, current time: 4
mbedtls: ssl_cli.c:777 dumping 'client hello, random bytes' (32 bytes)
mbedtls: ssl_cli.c:777 0000:  00 00 00 04 8e 4f c0 13 4d 9d 81 67 c1 d2 fb 01  .....O..M..g....
mbedtls: ssl_cli.c:777 0010:  9c ef 39 f9 a8 32 ae b2 5c ce ec d0 9a f6 08 84  ..9..2..\.......
mbedtls: ssl_cli.c:830 client hello, session id len.: 0
mbedtls: ssl_cli.c:831 dumping 'client hello, session id' (0 bytes)
mbedtls: ssl_cli.c:898 client hello, add ciphersuite: 003d
mbedtls: ssl_cli.c:898 client hello, add ciphersuite: 0035
mbedtls: ssl_cli.c:898 client hello, add ciphersuite: 003c
mbedtls: ssl_cli.c:898 client hello, add ciphersuite: 002f
mbedtls: ssl_cli.c:905 client hello, got 4 ciphersuites (excluding SCSVs)
mbedtls: ssl_cli.c:914 adding EMPTY_RENEGOTIATION_INFO_SCSV
mbedtls: ssl_cli.c:963 client hello, compress len.: 1
mbedtls: ssl_cli.c:965 client hello, compress alg.: 0
mbedtls: ssl_cli.c:72 client hello, adding server name extension: abcdefghsn-ats.iot.us-east-1.amazonaws.com
mbedtls: ssl_cli.c:189 client hello, adding signature_algorithms extension
mbedtls: ssl_cli.c:555 client hello, adding extended_master_secret extension
mbedtls: ssl_cli.c:1039 client hello, total extension length: 75
mbedtls: ssl_tls.c:2749 => write record
mbedtls: ssl_tls.c:2895 output record: msgtype = 22, version = [3:1], msglen = 130
mbedtls: ssl_tls.c:2898 dumping 'output record sent to network' (135 bytes)
mbedtls: ssl_tls.c:2898 0000:  16 03 01 00 82 01 00 00 7e 03 03 00 00 00 04 8e  ........~.......
mbedtls: ssl_tls.c:2898 0010:  4f c0 13 4d 9d 81 67 c1 d2 fb 01 9c ef 39 f9 a8  O..M..g......9..
mbedtls: ssl_tls.c:2898 0020:  32 ae b2 5c ce ec d0 9a f6 08 84 00 00 0a 00 3d  2..\...........=
mbedtls: ssl_tls.c:2898 0030:  00 35 00 3c 00 2f 00 ff 01 00 00 4b 00 00 00 33  .5.<./.....K...3
mbedtls: ssl_tls.c:2898 0040:  00 31 00 00 2e 61 31 6d 69 67 39 30 6b 39 62 6d  .1...abcdefghsn
mbedtls: ssl_tls.c:2898 0050:  6d 63 76 2d 61 74 73 2e 69 6f 74 2e 75 73 2d 65  mcv-ats.iot.us-e
mbedtls: ssl_tls.c:2898 0060:  61 73 74 2d 31 2e 61 6d 61 7a 6f 6e 61 77 73 2e  ast-1.amazonaws.
mbedtls: ssl_tls.c:2898 0070:  63 6f 6d 00 0d 00 0c 00 0a 06 01 05 01 04 01 03  com.............
mbedtls: ssl_tls.c:2898 0080:  01 02 01 00 17 00 00                             .......
mbedtls: ssl_tls.c:2464 => flush output
mbedtls: ssl_tls.c:2483 message length: 135, out_left: 135
mbedtls: ssl_tls.c:2489 ssl->f_send() returned 135 (-0xffffff79)
mbedtls: ssl_tls.c:2508 <= flush output
mbedtls: ssl_tls.c:2907 <= write record
mbedtls: ssl_cli.c:1065 <= write client hello
mbedtls: ssl_cli.c:3336 client state: 2
mbedtls: ssl_tls.c:2464 => flush output
mbedtls: ssl_tls.c:2476 <= flush output
mbedtls: ssl_cli.c:1458 => parse server hello
mbedtls: ssl_tls.c:3785 => read record
mbedtls: ssl_tls.c:2256 => fetch input
mbedtls: ssl_tls.c:2414 in_left: 0, nb_want: 5
mbedtls: ssl_tls.c:2438 in_left: 0, nb_want: 5
mbedtls: ssl_tls.c:2439 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
mbedtls: ssl_tls.c:2451 <= fetch input
mbedtls: ssl_tls.c:3535 dumping 'input record header' (5 bytes)
mbedtls: ssl_tls.c:3535 0000:  16 03 03 13 94                                   .....
mbedtls: ssl_tls.c:3544 input record: msgtype = 22, version = [3:3], msglen = 5012
mbedtls: ssl_tls.c:2256 => fetch input
mbedtls: ssl_tls.c:2414 in_left: 5, nb_want: 5017
mbedtls: ssl_tls.c:2438 in_left: 5, nb_want: 5017
mbedtls: ssl_tls.c:2439 ssl->f_recv(_timeout)() returned 4351 (-0xffffef01)
mbedtls: ssl_tls.c:2438 in_left: 4356, nb_want: 5017
mbedtls: ssl_tls.c:2439 ssl->f_recv(_timeout)() returned 661 (-0xfffffd6b)
mbedtls: ssl_tls.c:2451 <= fetch input
mbedtls: ssl_tls.c:3714 dumping 'input record from network' (5017 bytes)
mbedtls: ssl_tls.c:3714 0000:  16 03 03 13 94 02 00 00 51 03 03 64 00 fe 83 53  ........Q..d...S
mbedtls: ssl_tls.c:3714 0010:  fc aa fb 6e 17 ca a8 aa 14 c4 f5 e6 3f 47 cb e2  ...n........?G..
mbedtls: ssl_tls.c:3714 0020:  f7 02 62 9d 85 1a f1 0d 40 8d bd 20 08 19 6e 5a  ..b.....@.. ..nZ
[... 866 lines cut ...]
mbedtls: ssl_tls.c:692 0080:  b9 15 8b 6f 75 a4 33 0a d2 bb 76 b3 33 1f 19 ef  ...ou.3...v.3...
mbedtls: ssl_tls.c:692 0090:  96 17 08 40 42 7f ae 56 6b e4 14 78 95 eb a5 32  ...@B..Vk..x...2
mbedtls: ssl_tls.c:692 00a0:  ce f4 17 8c 45 44 4f d8 e8 8b 25 2b 67 85 67 da  ....EDO...%+g.g.
mbedtls: ssl_tls.c:692 00b0:  f4 1b 0a e6 4d 57 65 64 78 b7 1c 05 5b 48 f7 35  ....MWedx...[H.5
mbedtls: ssl_tls.c:692 00c0:  10 00 ec f2 22 da c4 3f 9a 02 14 88 53 f6 aa db  ...."..?....S...
mbedtls: ssl_tls.c:692 00d0:  6a de 5c 51 84 6d 4e 35 5e b4 25 cc 99 b7 bf 1a  j.\Q.mN5^.%.....
mbedtls: ssl_tls.c:692 00e0:  87 62 da de 3e c8 81 da 03 0c 84 df 82 c1 53 b6  .b..>.........S.
mbedtls: ssl_tls.c:692 00f0:  d0 de c2 50 8d 22 6e 91 04 bd 87 3e 96 a5 0d 79  ...P."n....>...y
mbedtls: ssl_tls.c:800 keylen: 16, minlen: 64, ivlen: 16, maclen: 32
mbedtls: ssl_tls.c:990 <= derive keys
mbedtls: ssl_tls.c:1079 => calc verify sha256
mbedtls: ssl_tls.c:1084 dumping 'calculated verify result' (32 bytes)
[... cut ...]
mbedtls: ssl_tls.c:1084 0010:  41 70 86 39 bb 36 1a a5 b9 a0 55 6a 29 07 58 3e  Ap.9.6....Uj).X>
mbedtls: ssl_tls.c:1085 <= calc verify
mbedtls: ssl_cli.c:3190 mbedtls_pk_sign() returned -17168 (-0x4310)
mbedtls: ssl_tls.c:6684 <= handshake
E (17380) esp-tls: mbedtls_ssl_handshake returned -0x4310
E (17400) esp-tls: Failed to open new connection
E (17400) TRANS_SSL: Failed to open a new connection
E (17410) MQTT_CLIENT: Error transport connect
I (17410) MQTT_CLIENT: Reconnect after 10000 ms
`

@donghengqaz please find the logs for the mbedtls handshake error.

donghengqaz, 2019-11-07:
I have checked your log and found that the resean of your problem is that there is not enough memory for you to create a SSL connection, so the `-0x4310` appears.

So I guess after the SSL connect breaks, you may not free some memory resource, you may try to find those place and do some things.

sky9971, 2019-11-08:
Yes that's correct, I also came to the same conclusion that there isn't enough memory for ssl to reconnect.

Can you help me understand how can I free memory at this point when ssl connection breaks?
In my firmware i'm using esp http server, esp mqtt client and a udp echo server(1kb).

sky9971, 2019-11-09:
@donghengqaz 
I noticed that before ESP mqtt library tries to reconnect  it only calls **esp_transport_close(esp_transport_handle_t t)** but it does not free the memory acquired by the transport handle.

Would be a good idea to replace/also use this method with **esp_transport_destroy(esp_transport_handle_t t)**
at this block https://github.com/espressif/ESP8266_RTOS_SDK/blob/9041579f4a6877ff33ac8f752c94cb5b93d9f71a/components/mqtt/esp-mqtt/mqtt_client.c#L746

so that we clean up everything before retrying

sky9971, 2019-11-18:
@donghengqaz any update on this?
Is there any other way/alternate to overcome this?

I tried the ESP MQTT library with WolfSSL which seems to give me a min heap memory of 40k.
But it does not reconnect and gets stuck in **esp-tls: handshake in progress message**

I enabled debug for WOLFSSL which showed that the **WOLFSSL is encountering an error -323 during reconnection**.

swetha-paladugu-oyo, 2020-09-14:
@sky9971 Is your issue fixed?
i am also facing the same issue, same error code -0x4310 while mqtt reconnection.
I have added few logs in tls lower layers to find out where memory is getting consumed and i understood during connection memory is getting consumed and memory should be freed when connection terminates or when failed to connect.
so in such cases internally "esp_tls_conn_delete" api  in esp_tls.c is called to free the memory. But what i observed is that the minimum heap available is same before and after calling this "esp_tls_conn_delete" function.

void esp_tls_conn_delete(esp_tls_t *tls)
{
    if (tls != NULL) {
        esp_tls_cleanup(tls);
#if CONFIG_SSL_USING_MBEDTLS
        if (tls->is_tls) {
            mbedtls_net_free(&tls->server_fd);
        } else
#endif
        if (tls->sockfd >= 0) {
            close(tls->sockfd);
        }
        free(tls);
        ESP_LOGW(TAG, "esp_get_minimum_free_heap_size %d bytes", esp_get_minimum_free_heap_size());

    }
};

Can anyone please tell why this free is not working?

sky9971, 2020-09-16:
@swetha-paladugu-oyo 
Yes my issue is resolved now. I found it more of as how the firmware is designed or these MQTT SSL modules are implemented in your firmware then as an issue in the SDK. So my suggestion would be to go back to firmware code and identify where you are allocating memory, how are you allocating memory whether statically or dynamically and whether you are freeing memory after the use or not. You need not make any changes in the esp_tls.c

If you are in need of professional help in resolving this issue, I can also offer technical support to fix the issue.
You may contact me at akash@mansaa.io

swetha-paladugu-oyo, 2020-09-16:
@sky9971 
Hi, In my code no where i am allocating memory dynamically... After seeing free() issue in esp_tls.c file, in my main() i tried allocating 16k memory and freeing it... but here also free is not happening.. the minimum heap size available remains same before and after calling free... I strongly believe free() is not working..

sky9971, 2020-09-16:
@swetha-paladugu-oyo 
I will still recommend going back to your code and identifying where you are allocating memory and whether it is being freed or not. If you find that you have freed all the dynamically allocated memory then update your SDK to the latest stable version and then try again.
One last suggestion from my end will be to use dynamic memory allocation instead of statically assigning values or memory to your variables. 
I am afraid I won't be able to provide any further assistance beyond this point for free. However I am still open to professional support to resolve your issue.