espressif/ESP8266_RTOS_SDK #749: Unable to connect to mqtt server over ssl

sreeisalso, 2019-11-02:
Hi I'm unable to connect to mqtt server hosted using mosquitto 1.6.5

esp8266 log 
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 6120, room 16 
tail 8
chksum 0xa3
load 0x3ffe8408, len 24, room 0 
tail 8
chksum 0x2a
load 0x3ffe8420, len 3460, room 0 
tail 4
chksum 0xb0
I (46) boot: ESP-IDF v3.2-222-g2b299ac7-dirty 2nd stage bootloader
I (46) boot: compile time 21:36:06
I (47) qio_mode: Enabling default flash chip QIO
I (55) boot: SPI Speed      : 40MHz
I (61) boot: SPI Mode       : QIO
I (68) boot: SPI Flash Size : 2MB
I (74) boot: Partition Table:
I (79) boot: ## Label            Usage          Type ST Offset   Length
I (91) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (114) boot:  2 factory          factory app      00 00 00010000 000f0000
I (125) boot: End of partition table
I (132) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x5dcb0 (384176) map
0x40210010: _stext at ??:?

I (283) esp_image: segment 1: paddr=0x0006dcc8 vaddr=0x4026dcc0 size=0x11b90 ( 72592) map
I (310) esp_image: segment 2: paddr=0x0007f860 vaddr=0x3ffe8000 size=0x005e8 (  1512) load
I (311) esp_image: segment 3: paddr=0x0007fe50 vaddr=0x40100000 size=0x00a50 (  2640) load
I (322) esp_image: segment 4: paddr=0x000808a8 vaddr=0x40100a50 size=0x054e0 ( 21728) load
I (342) boot: Loaded app from partition at offset 0x10000
D (354) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
I (373) system_api: Base MAC address is not set, read default base MAC address from EFUSE
D (382) nvs: nvs_open_from_partition backup_mac 0
D (387) nvs: nvs_get_str_or_blob backup_mac_data
D (393) nvs: nvs_close 1
I (396) system_api: Base MAC address is not set, read default base MAC address from EFUSE
D (405) nvs: nvs_open_from_partition backup_mac 0
D (410) nvs: nvs_get_str_or_blob backup_mac_data
D (416) nvs: nvs_close 2
D (419) phy_init: loading PHY init data from application binary
D (425) nvs: nvs_open_from_partition phy 0
D (430) nvs: nvs_get_str_or_blob cal_data
D (435) nvs: nvs_get_str_or_blob dc_table
D (442) nvs: nvs_close 3
phy_version: 1153.0, 0398f21, Oct 24 2019, 20:29:07, RTOS new
I (492) phy_init: phy ver: 1153_0
I (497) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
D (504) wdt: Enable task watch dog panic, panic time parameter is 11
D (511) wdt: task watch dog trigger time parameter is 15
I (517) MQTTS_EXAMPLE: [APP] Startup..
I (522) MQTTS_EXAMPLE: [APP] Free memory: 98520 bytes
I (531) MQTTS_EXAMPLE: [APP] IDF version: v3.2-222-g2b299ac7-dirty
D (546) nvs: nvs_open_from_partition misc 1
D (551) nvs: nvs_open_from_partition nvs.net80211 1
D (557) nvs: nvs_get opmode 1
D (561) nvs: nvs_get_str_or_blob sta.ssid
D (566) nvs: nvs_get_str_or_blob sta.mac
D (571) nvs: nvs_get sta.authmode 1
D (575) nvs: nvs_get_str_or_blob sta.pswd
D (581) nvs: nvs_get_str_or_blob sta.pmk
D (586) nvs: nvs_get sta.chan 1
D (591) nvs: nvs_get auto.conn 1
D (595) nvs: nvs_get bssid.set 1
D (600) nvs: nvs_get_str_or_blob sta.bssid
D (606) nvs: nvs_get sta.lis_intval 2
[... 286 lines cut ...]

W (6146) mbedtls: ssl_tls.c:4344 ssl_get_next_record() returned -80 (-0x0050)

W (6158) mbedtls: ssl_cli.c:1506 mbedtls_ssl_read_record() returned -80 (-0x0050)

I (6170) mbedtls: ssl_tls.c:8094 <= handshake

E (6178) esp-tls: mbedtls_ssl_handshake returned -0x50
E (6189) esp-tls: Failed to open new connection
E (6195) TRANS_SSL: Failed to open a new connection
E (6203) MQTT_CLIENT: Error transport connect
I (6211) MQTT_CLIENT: Reconnect after 10000 ms
I (6219) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
```

mosquitto 1.6.5 server log
```
1572710899: New connection from 117.99.80.128 on port 8883.
1572710900: Client <unknown> disconnected due to protocol error.
```

I used  code from example folder without any modification
```
~/esp/ESP8266_RTOS_SDK/examples/protocols/esp-mqtt/ssl/
```

donghengqaz, 2019-11-04:
I see your log and find that you send a SSL client hello message, but the SSL server close the connection, so I guess if the protocol of the client is not supported by the server. I don't know how you configurate the SSL server, so I can't give some direct sollution.

If you have learned some SSL knowalage, you can configurate the SSL at "menuconfig" by your self, another way is that you can try the master branch, I updated the mbedTLS and its default supported function.

OskOnu, 2019-11-27:
Did you solved the issue?

closed 2020-09-25 by sreeisalso