espressif/ESP8266_RTOS_SDK #750: Cannot read CCOUNT on master branch

DKrepsky, 2019-11-03:
INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF version: 2b299ac7
- Development Env:      [Make|CMake]
- Operating System:     [Ubuntu]
- Power Supply:         [external 5V]

## Problem Description

The value of ccount register is not being updated.

### Expected Behavior

The value of ccount increases for every clock pulse.

### Actual Behavior

Reading ccount always returns the same value, both with xthal_get_ccount() and through asm.
Same come works on branch release/v3.2

### Steps to reproduce

1- Build sample code with current master branch.
2- Flash and monitor.
3- Build same sample code with branch release/v3.2.
4- Flash and monitor.

[ccount.elf.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3802327/ccount.elf.zip)
Module ESP-12F, no external components required.

### Code to reproduce this issue

```cpp
        #include <stdio.h>
        #include "freertos/FreeRTOS.h"
        #include "freertos/task.h"
        #include "xtensa/hal.h"

        void app_main()
        {
            while(1){
                vTaskDelay(pdMS_TO_TICKS(1000));
                printf("%ul\r\n", xthal_get_ccount());
            }
        }
```
## Debug Logs

```
I (44) boot: ESP-IDF v3.2-222-g2b299ac7 2nd stage bootloader
I (44) boot: compile time 20:51:27
I (44) qio_mode: Enabling default flash chip QIO
I (45) boot: SPI Speed      : 40MHz
I (45) boot: SPI Mode       : QIO
I (45) boot: SPI Flash Size : 4MB
I (45) boot: Partition Table:
I (45) boot: ## Label            Usage          Type ST Offset   Length
I (46) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (46) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (47) boot:  2 factory          factory app      00 00 00010000 000f0000
I (47) boot: End of partition table
I (47) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x1f354 (127828) map
0x40210010: _flash_cache_start at ??:?

I (91) esp_image: segment 1: paddr=0x0002f36c vaddr=0x4022f364 size=0x04138 ( 16696) map
I (97) esp_image: segment 2: paddr=0x000334ac vaddr=0x3ffe8000 size=0x0055c (  1372) load
I (98) esp_image: segment 3: paddr=0x00033a10 vaddr=0x40100000 size=0x008c0 (  2240) load
I (99) esp_image: segment 4: paddr=0x000342d8 vaddr=0x401008c0 size=0x051c4 ( 20932) load
I (107) boot: Loaded app from partition at offset 0x10000
I (137) system_api: Base MAC address is not set, read default bse MAC address from EFUSE
I (141) system_api: Base MAC address is not set, read default base MAC address from EFUSE
phy_version: 1153.0, 0398f21, Oct 24 2019, 20:29:07, RTOS new
I (184) phy_init: phy ver: 1153_0
I (186) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2
894l
894l
894l
894l
894l
894l
894l
894l
894l
894l
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [X] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

donghengqaz, 2019-11-04:
I am so sorry that I modified the "CPU" tick count using method, so it will be reset after CCOMPARE interrupt callback. And could you please tell me why you should use this? If you want to use micro second, you can call “esp_timer_get_time”.

DKrepsky, 2019-11-04:
Some of my company products needed very optimized assemble code running at 160MHz to avoid external signal conditioning. Using ccount did the trick and avoided a function call to get accurate timing, so we used it a lot for other things too.
I will migrate whatever I can to esp_timer_get_time and what can't be refactored will remain with v3.2.
Thanks for the fast response.
Best regards.

closed 2019-11-04 by DKrepsky