espressif/ESP8266_RTOS_SDK #751: ESP8266 跑example下的smartconfig  pannic

luweiqi, 2019-11-05:
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 7196, room 16 
tail 12
chksum 0x15
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x9c
ho 0 tail 12 room 4
load 0x3ffe8420, len 3548, room 12 
tail 0
chksum 0xc2
I (95) boot: ESP-IDF v3.2-218-g752fdc4-dirty 2nd stage bootloader
I (95) boot: compile time 11:08:46
I (97) qio_mode: Enabling default flash chip QIO
I (113) boot: SPI Speed      : 40MHz
I (126) boot: SPI Mode       : QIO
I (139) boot: SPI Flash Size : 2MB
I (151) boot: Partition Table:
I (163) boot: ## Label            Usage          Type ST Offset   Length
I (185) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (209) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (232) boot:  2 factory          factory app      00 00 00010000 000f0000
I (255) boot: End of partition table
I (268) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x41764 (268132) map
0x40210010: _flash_cache_start at ??:?

I (423) esp_image: segment 1: paddr=0x0005177c vaddr=0x40251774 size=0x07814 ( 30740) map
I (438) esp_image: segment 2: paddr=0x00058f98 vaddr=0x3ffe8000 size=0x005ac (  1452) load
I (445) esp_image: segment 3: paddr=0x0005954c vaddr=0x40100000 size=0x00a30 (  2608) load
I (473) esp_image: segment 4: paddr=0x00059f84 vaddr=0x40100a30 size=0x05564 ( 21860) load
I (509) boot: Loaded app from���ѥѥ���at offset 0x10000
I (538) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (546) system_api: Base MAC address is not set, read default base MAC address from EFUSE
I (608) phy_init: phy ver: 1152_0
I (610) reset_reason: RTC reset 2 wakeup 0 store 3, reason is 2
SC version: V2.8.2
I (4161) sc: SC_STATUS_FINDING_CHANNEL
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 0 register dump:
PC      : 0x40101379  PS      : 0x00000033  A0      : 0x40101303  A1      : 0x3fff1c70  
0x40101379: _heap_caps_free at /mnt/hgfs/ubuntu_esp/esp/ESP8266_RTOS_SDK/components/heap/src/esp_heap_caps.c:277

0x40101303: blk_link_size at /mnt/hgfs/ubuntu_esp/esp/ESP8266_RTOS_SDK/components/heap/port/esp8266/include/priv/esp_heap_caps_priv.h:100
 (inlined by) _heap_caps_free at /mnt/hgfs/ubuntu_esp/esp/ESP8266_RTOS_SDK/components/heap/src/esp_heap_caps.c:255

A2      : 0x005e005e  A3      : 0x80000000  A4      : 0x005e005e  A5      : 0x3ffeae68  
A6      : 0x005e005e  A7      : 0x005e005e  A8      : 0x000000ff  A9      : 0x000000b0  
A10     : 0x00000001  A11     : 0xffffffff  A12     : 0x3ffeae68  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x0000001d  EXCCAUSE: 0x00000009  

��xxx�xxx�x�xxx�x��x�xx�x��x�x�x�xx�x�x�x�x����x�x�xxx����xx�x��xx��x�x�x�x�x�xx�x�x�x�x�xxxx�x��x�x�x��xxx�x�x�x�x�xxxxxxx�x��x�x�x���xx�x�x�x�x�xx�x�x�xxx�xxxx�x�x�x�xxx�����xxx�x��x�x�xxx�x��xxx��xx��x�xx�x�����������x���x��x�x��xx��x��x�����������x��������������xx������������x���x��x�x�����x��xx������������x����x�xxxx�����xx�xxx�������������x����x�x�x�x�x�x�x��xx�������������x����x�xxxxxxxx��xx�������������x��x��x�x��x��x�x��xxx������������x����x�xxxx�x����xx�xxx�������������x����������xx�������������x����x�xxxxxxxx��xx������������x��x��x�x��x��x�x��xx�������������xx�x����xx�x�xx����xx�xxx������������x�������x���xx�x�����������x����������x�xxxx��xx��������������x�������x���xx��������������x�����������xx�xxx�������������x����x�xxxxxxx����xx�x������������x���������������xxx�x����������x������x�����xxx�xxxxx��xxx����x�������x���xx�xxx�x�xxxxx�x�xxxxx����x���x��x�x��xx���x����x�xxxx�����x���x��x����x�����x����x�xxxx�������x���x�����x�������x����x�xxxx�xx��xx�
0x40101379: _heap_caps_free at /mnt/hgfs/ubuntu_esp/esp/ESP8266_RTOS_SDK/components/heap/src/esp_heap_caps.c:277

0x4024eaf2: ppProcessRxPktHdr at ??:?

0x4024fdbe: ppTask at ??:?

donghengqaz, 2019-11-07:
Developers in my side have found the problem, and we are trying our best to fix it, it is done, I will notify you in the issue.

luweiqi, 2019-11-08:
hello donghengqaz,is this problem happened in the  earlier version?  By the way, would you please offer me a practicable  version?  Thanks very much.

FireWinter, 2020-01-07:
Please try the sdk release/v3.3 @luweiqi

angggh, 2020-02-13:
ESP8285 出现同样的问题