espressif/ESP8266_RTOS_SDK #752: Did you test it with 40MHz crystal?

jandrassy, 2019-11-05:
## Environment

- Development Kit: Arduino Uno WiFi Developer Edition (ESP8266 with 40 MHz crystal)
- IDF version git a59b346
- Development Env:     Make
- Operating System:    Ubuntu
- Power Supply:         USB

## Problem Description

I can't make the esp-at firmware or the IDF examples work on esp8266 with 40 MHz crystal. I set the 40 MHz crystal in menuconfig. The hello_world example and simple_wifi examples print at 115200 baud only if in menuconfig "Component config > ESP8266-specific > UART console baud rate" is set to 74880. The simple_wifi example doesn't connect to AP to which NONOS SDK on same board connects without problems. And the AT firmware responses only garbage on UART.

[sdkconfig.zip](https://github.com/espressif/ESP8266_RTOS_SDK/files/3810846/sdkconfig.zip)

AT firmware output

```
Connect to serial port /dev/ttyACM0 at 115200

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 6252, room 16  
tail 12
chksum 0x94
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x7d
ho 0 tail 12 room 4
load 0x3ffe8420, len 3452, room 12 
tail 0
chksum 0xe1
csum 0xe1
`J��
>>Send to /dev/ttyACM0: "AT"<<
���
```
simple_wifi

```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 7196, room 16  
tail 12
chksum 0xcf
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x26
ho 0 tail 12 room 4
load 0x3ffe8420, len 3544, room 12 
tail 12
chksum 0x7d
csum 0x7d
[0;32mI (34) boot: ESP-IDF v3.2-220-ga59b346 2nd stage bootloader[0m
[0;32mI (34) boot: compile time 20:27:43[0m
[0;32mI (34) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (39) boot: SPI Speed      : 40MHz[0m
[0;32mI (43) boot: SPI Mode       : QIO[0m
[0;32mI (47) boot: SPI Flash Size : 4MB[0m
[0;32mI (51) boot: Partition Table:[0m
[0;32mI (55) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m 
[0;32mI (77) boot:  2 factory          factory app      00 00 00010000 000f0000[0m
[0;32mI (84) boot: End of partition table[0m
[0;32mI (88) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x39c6c (236652) map[0m
[0;32mI (148) esp_image: segment 1: paddr=0x00049c84 vaddr=0x40249c7c size=0x07008 ( 28680) map[0m
[0;32mI (155) esp_image: segment 2: paddr=0x00050c94 vaddr=0x3ffe8000 size=0x0057c (  1404) load[0m
[0;32mI (156) esp_image: segment 3: paddr=0x00051218 vaddr=0x40100000 size=0x00a30 (  2608) load[0m
[0;32mI (165) esp_image: segment 4: paddr=0x00051c50 vaddr=0x40100a30 size=0x055cc ( 21964) load[0m
[0;32mI (178) boot: Loaded app from partition at offset 0x10000[
[0;32mI (194) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
phy_version: 1153.0, 0398f21, Oct 24 2019, 20:29:07, RTOS new
[0;32mI (258) phy_init: phy ver: 1153_0[0m
[0;32mI (260) reset_reason: RTC reset 1 wakeup 0 store 0, reason is 1[0m
[0;32mI (265) simple wifi: ESP_WIFI_MODE_STA[0m
[0;32mI (303) simple wifi: wifi_init_sta finished.[0m
[0;32mI (306) simple wifi: connect to ap SSID:X password:Y[0m
[0;31mE (2753) simple wifi: Disconnect reason : 201[0m
[0;31mE (5200) simple wifi: Disconnect reason : 201[0m
```

jandrassy, 2019-11-06:
everything works if I change in phy_init_data.h the 48th byte to 0.

```
[0;32mI (223) simple wifi: ESP_WIFI_MODE_STA[0m
[0;32mI (256) simple wifi: wifi_init_sta finished.[0m
[0;32mI (259) simple wifi: connect to ap SSID:X password:Y[0m
[0;32mI (387) wifi: state: 0 -> 2 (b0)[0m
[0;32mI (391) wifi: state: 2 -> 3 (0)[0m
[0;32mI (396) wifi: state: 3 -> 5 (10)[0m
[0;32mI (398) wifi: pm start, type: 2[0m 
[0;32mI (3251) event: sta ip: 192.168.1.102, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (3256) simple wifi: got ip:192.168.1.102[0m 
```

```
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 6252, room 16 
tail 12
chksum 0xa9
ho 0 tail 12 room 4
load 0x3ffe8408, len 24, room 12 
tail 12
chksum 0x40
ho 0 tail 12 room 4
load 0x3ffe8420, len 3460, room 12 
tail 8
chksum 0x7b
csum 0x7b

ready

>>Send to /dev/ttyACM0: "AT"<<
AT

OK
```

where in build process should the 40MHz setting be applied over phy_init to partitions_singleapp.bin or partitions_at.bin?

donghengqaz, 2019-11-07:
I have planed to add the 40MHz STAL support for next version v3.4，because it is a hard work. I should check and modify a lot of hardware initialization code both for bootloader and application.

If you want to apply it the phy initialization, you may modify the bytes[48](we count it starting form bytes[0]) phy_init data from 1 to 0.

jandrassy, 2019-11-07:
> 
> 
> I have planed to add the 40MHz STAL support for next version v3.4，because it is a hard work. I should check and modify a lot of hardware initialization code both for bootloader and application.

I see

> If you want to apply it the phy initialization, you may modify the bytes[48](we count it starting form bytes[0]) phy_init data from 1 to 0.

exactly what I did. luckily the first line of the array is on line 30 so the flag is on 30 + 48 line 78

jandrassy, 2019-12-05:
so know you removed the option to select 40 MHz crystal?

jandrassy, 2019-12-09:
this seem to work with esp-at builds:
```
#if CONFIG_CRYSTAL_USED_40MHZ
        0x00,
#else
        0x01,
#endif
```

jandrassy, 2020-01-14:
please revert https://github.com/espressif/ESP8266_RTOS_SDK/commit/668c3bbb2c82a770a2afdb88a7f4440a0518b593
and the ifdef to  phy_init_data.h