espressif/ESP8266_RTOS_SDK #755: 怎样区分上次重启是由于断电、RSET、esp_restart()?

thisdream, 2019-11-07:
怎样区分上次重启是由于断电、RSET、esp_restart()?
----------------------------- Delete below -----------------------------
INSTRUCTIONS
============

无论是断电；REST引脚低电平；程序调用esp_restart()。调用esp_reset_reason()的结果均为ESP_RST_EXT。

但开机时设备自动输出log却有区别

断电重启：      store 3243988430
输出log为
(540) phy_init: phy ver: 1055_13
(540) reset_reason: RTC reset 2 wakeup 0 store 3243988430, reason is 2

REST引脚低电平：      store 0
输出log为
(540) phy_init: phy ver: 1055_13
(540) reset_reason: RTC reset 2 wakeup 0 store 0, reason is 2

程序调用esp_restart()：      store 3
输出log为
(540) phy_init: phy ver: 1055_13
(540) reset_reason: RTC reset 2 wakeup 0 store 3, reason is 2

但是不知道怎样读取store
----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF version: v3.2-20-g87e51aee-dirty
- Development Env:      vscode
- Operating System:     Windows
- Power Supply:         [USB|external 3.3V|Battery]

donghengqaz, 2019-11-14:
可以参考位于 "esp_system.h" 里面的 API: 
```
esp_reset_reason_t esp_reset_reason(void);
```

thisdream, 2019-11-21:
@donghengqaz   8266rtos的文档没有搜到esp_reset_reason_t的说明，参考了32rtos的说明。但是无论断电、RTC引脚、esp_restart导致的重启，调用esp_reset_reason的返回均为ESP_RST_EXT。代码如下，您也可以帮忙测试一下：
void app_main()
{
    esp_reset_reason_t temp = esp_reset_reason();
    switch (temp)
    {
    case ESP_RST_UNKNOWN:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_UNKNOWN");
        break;
    case ESP_RST_POWERON:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_POWERON");
        break;
    case ESP_RST_EXT:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_EXT");
        break;
    case ESP_RST_SW:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_SW");
        break;
    case ESP_RST_PANIC:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_PANIC");
        break;
    case ESP_RST_INT_WDT:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_INT_WDT");
        break;
    case ESP_RST_TASK_WDT:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_TASK_WDT");
        break;
    case ESP_RST_WDT:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_WDT");
        break;
    case ESP_RST_DEEPSLEEP:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_DEEPSLEEP");
        break;
    case ESP_RST_BROWNOUT:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_BROWNOUT");
        break;
    case ESP_RST_SDIO:
        TRACE_chenhuimed_Debug("reset_reason is ESP_RST_SDIO");
        break;
    }
    vTaskDelay(10000 / portTICK_RATE_MS);
    esp_restart();
}

XiaoSenLuo, 2020-06-12:
我也遇到了, 开发板是ESP-12F, 无论是按复位键还是断电, 重新运行都是 ESP_RST_EXT.