espressif/ESP8266_RTOS_SDK #756: Console input command causes memory leak

syj0925, 2019-11-13:
INSTRUCTIONS
============
Using example (console) in the esp8266 RTOS SDK to discover input command lines can cause dynamic memory leaks. In addition, if you keep querying the same command, memory leaks only once.

log1：
esp8266> 
esp8266> free
87152
esp8266> heap
I (10940) cmd_system: min heap size: 87000
esp8266> free
87120
esp8266> 

log2:
esp8266> 
esp8266> free
87120
esp8266> 
esp8266> tasks
Task Name	Status	Prio	HWM	Task#
uiT            	X	14	5848	1
IDLE           	R	0	848	2
Tmr Svc        	B	2	1872	3
esp8266> free
87088
esp8266> 

p.s. Memory fixed leak 32bytes
----------------------------- Delete above -----------------------------

## Environment

- Development Kit:
- IDF versionIDF Version:v3.2-252-g73bf28b-dirty
- Development Env:     Source insight
- Operating System:    Ubuntu
- Power Supply:         USB

donghengqaz, 2019-11-14:
The key maybe your said “memory leaks only once”. This it not a system that runs after preparing all resource. Some part of the functions  will "alloc" memory when is to use it. So the "lose" memory may just be the necessary resource for the console or its command.

syj0925, 2019-11-19:
Maybe I didn't describe it clearly enough. I mean, if you keep typing the same command, memory leaks only once. However, if two different commands are cycled, memory will always leak.