espressif/ESP8266_RTOS_SDK #764: 更新 sdk 之后 make menuconfig 问题 ‘python\r’: No such file or directory ??

panguoxian, 2019-11-23:
make menuconfig如下问题

/usr/bin/env: ‘python\r’: No such file or directory
make[1]: /media/share/esp8266_RTOS_SDK/tools/kconfig/lxdialog/check-lxdialog.sh: Command not found
make[1]: Entering directory `/media/share/esp8266_RTOS_SDK/tools/kconfig'
cc -c   -DLOCALE -MMD -MP -I "." -I "/media/share/esp8266_RTOS_SDK/tools/kconfig"  /media/share/esp8266_RTOS_SDK/tools/kconfig/mconf.c -o mconf.o
In file included from /media/share/esp8266_RTOS_SDK/tools/kconfig/mconf.c:23:
/media/share/esp8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:38: error: #include expects "FILENAME" or <FILENAME>
/media/share/esp8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:103: error: expected specifier-qualifier-list before ‘chtype’
/media/share/esp8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:200: error: expected ‘)’ before ‘*’ token
/media/share/esp8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:221: error: expected ‘)’ before ‘*’ token
/media/share/esp8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:223: error: expected ‘)’ before ‘*’ token
/media/share/esp8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:224: error: expected ‘)’ before ‘*’ token
/media/share/esp8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:225: error: expected ‘)’ before ‘*’ token
/media/share/esp8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:226: error: expected ‘)’ before ‘*’ token
/media/share/esp8266_RTOS_SDK/tools/kconfig/lxdialog/dialog.h:228: error: expected ‘)’ before ‘*’ token
/media/share/esp8266_RTOS_SDK/tools/kconfig/mconf.c: In function ‘conf’:
/media/share/esp8266_RTOS_SDK/tools/kconfig/mconf.c:671: error: ‘KEY_MAX’ undeclared (first use in this function)
/media/share/esp8266_RTOS_SDK/tools/kconfig/mconf.c:671: error: (Each undeclared identifier is reported only once
/media/share/esp8266_RTOS_SDK/tools/kconfig/mconf.c:671: error: for each function it appears in.)
/media/share/esp8266_RTOS_SDK/tools/kconfig/mconf.c: In function ‘show_help’:
/media/share/esp8266_RTOS_SDK/tools/kconfig/mconf.c:793: error: ‘stdscr’ undeclared (first use in this function)
/media/share/esp8266_RTOS_SDK/tools/kconfig/mconf.c: In function ‘conf_choice’:
/media/share/esp8266_RTOS_SDK/tools/kconfig/mconf.c:855: error: ‘KEY_MAX’ undeclared (first use in this function)
make[1]: *** [mconf.o] Error 1
make[1]: Leaving directory `/media/share/esp8266_RTOS_SDK/tools/kconfig'
make: *** [/media/share/esp8266_RTOS_SDK/tools/kconfig/conf-idf] Error 2

panguoxian, 2019-11-23:
已解决

hawkwong, 2019-11-25:
本人也遇到同样问题，请问是如何解决的？

chunyexixiaoyu, 2020-12-24:
> 本人也遇到同样问题，请问是如何解决的？

我也遇到了如何解决的

Reim5, 2024-05-13:
If you guys are using a Debian-based distro, you can try installing `python-is-python3` to resolve this issue.

```(bash)
$ sudo apt install python-is-python3
```